RobotWebTools/rclnodejs #528: Max supported node/npm version not specified, leads to install errors

wayneparrott, 2019-10-28:
While the README specifies the lowest supported version of node it does not specify the highest supported version of node.

Recently while getting familiar with rclnodejs, I immediately encountered an error when attempting to install the package (v0.10.2) with the latest versions of node and npm, v13.0 and v6.12 respectively.  I systematically downgraded node and npm until I was able to install rclnodejs with node 10.17.9 and npm 6.11.3.

Propose updating the README and package.json to list 10.17.0 as the max supported version. I will be glad to submit a PR with these small but time saving changes if you agree.

minggangw, 2019-10-29:
>Propose updating the README and package.json to list 10.17.0 as the max supported version. I will be glad to submit a PR with these small but time saving changes if you agree.

Of course, please go ahead to submit and you can add yourself as a contributor (https://github.com/RobotWebTools/rclnodejs#contributors) simultaneously if you like, thanks again! 

Thanks for your finding! I also tested with the latest Erbium (v12.13.0) and had the same [result](https://github.com/RobotWebTools/rclnodejs/pull/527), and I found the error is caused by the [ref](https://www.npmjs.com/package/ref) which the rclnodejs depends on heavily (I reckon this is a problem in the future, and the ref seems to be out of maintenance).

At the beginning of this project, I only indicated the minimum version of node.js, because we used some ES6 features which are not supported by the lower ones. As for now, I think we'd better point out the maximum version to avoid the confusion.

minggangw, 2019-10-30:
Close this issue, as 90ebf88b1298258c33ed290c5c2a030b0be9b312 has landed.

closed 2019-10-30 by minggangw