RobotWebTools/rclnodejs #534: best resources for users

wayneparrott, 2019-11-12:
What is the recommended resource(s) for rclnodejs users to ask questions and share information?

I'm aware of the subreddit https://www.reddit.com/r/ROS/

Are there any other resources we should consider using, e.g., stackoverflow, discuss, discourse, etc?

Wayne
PS - Thx for enabling the JavaScript coders to participate in the ROS2 developer community.

minggangw, 2019-11-13:
I recommand https://discourse.ros.org/c/ng-ros which is the official forum of ROS community, you could communicate with other developers if you have problems or you can share your ideas or announce something you want others to know.

The downside is that this forum is not rclnodejs specific, but I think it's still a suitable place where you can get help, thanks!

wayneparrott, 2019-11-13:
Thx. Short summary:

https://discourse.ros.org/c/ng-ros 
http://planet.ros.org/
https://www.reddit.com/r/ROS/