RobotWebTools/rclnodejs #537: support TypeScript

wayneparrott, 2019-11-19:
I recognize and appreciate the commitment to ES6 by the project team. But I primarily favor TypeScript over JavaScript for much of my JS development. To that I recently hacked together a preliminary set of TypeScript typed definition (*.d.ts) files for the rclnodejs api for my own use. If there is interest I would gladly contribute this working to the rclnodejs project and support them going forward as the api evolves. See https://www.npmjs.com/package/@wayneparrott/rclnodejs-types
Alternatively, https://github.com/DefinitelyTyped/DefinitelyTyped could be a destination for the typedefs but personally I find projects that include integrated TS support feel more inclusive for TS developers and I would like to coordinate such efforts through this project before going outside it.

minggangw, 2019-11-20:
Wow, fantastic work! I still remember when I presented `rclnodejs` on ROSCon2018 in Madrid last year, I was asked whether the `rclnodejs` supports TypeScript, I think we can say that we have a solution that can use TypeScript now :smile:  Some comments:

> but personally I find projects that include integrated TS support feel more inclusive for TS developers and I would like to coordinate such efforts through this project before going outside it.

I agree it's more convenient for developers if we could integrate the TypeScript support inside `rclnodejs`, we can land the TS here first. Some work I can think out

-  How to integrate the TS into the current project, I suppose developers can select between JS and TS after the change
- How about the unit test, are we going to add some tests designed for TS interfaces?
- The document is going to be updated, we have [on-line](http://robotwebtools.org/rclnodejs/docs/index.html) version and the README

Finally, I think we can use this issue to track the following progress if you like, thanks again!

wayneparrott, 2019-11-21:
> How to integrate the TS into the current project, I suppose developers can select between JS and TS after the change

Correct. 
To get started I recommend that we manually maintain ts declaration files (*.d.ts) separate from the js api in /lib. For example maintain them in a directory named "types" or something similar. Once setup it takes very little time to maintain as the api evolves.

*Initially I tried using a jsdoc->*.d.ts conversion utility but it failed miserably. I hacked up the current set of types in a couple of hours, about the same about of time I spent trying to get the utility to work correctly and create something usable.*

Additionally, modify package.json to include a "types" property pointing to ./types/index.d.ts and extend the eslint configuration to include typescript support. 

> How about the unit test, are we going to add some tests designed for TS interfaces?

Because declaration files are design-time only artifacts and there is no new runtime code we could consider using a tool such as [dtslint](https://github.com/microsoft/dtslint) or [tsd](https://github.com/SamVerschueren/tsd). This is similar to the definitelytyped testing guidance.

> The document is going to be updated, we have on-line version and the README

If you want to keep the README about where it is now, consider breaking out new content for using TS with rclnodejs to a separate doc that is referenced by the README. I'll be glad to draft something for TS developers. 

As for TS api, perhaps we defer publishing additional api docs beyond the jsdoc for a while. A primary benefit of the ts declaration files will be when used with typescript coding tools for code assist and validation.

My next steps in prep for a contribution are:
1) complete a set of dstlint unit tests
2) update eslint rules to match those of rclnodejs with typescript configuration and ensure declaration files are valid
3) contribute a PR towards a new feature branch 

Thoughts?

wayneparrott, 2019-11-21:
fun times! The latest version of typescript is able to generate TS declaration files from js with jsdoc annotations.

While catching up on the latest typescript 3.7 features and I learned that the compiler (tsc) now supports compiler options that enable [using the "allowJs" and "declarations" options together](https://www.typescriptlang.org/docs/handbook/release-notes/typescript-3-7.html#--declaration-and---allowjs). In previous versions this was an error. What this means is the typescript compiler can now generate declaration files (*.d.ts) from js with jsdoc. 

I ran a quick test and ts declaration files were generated for all but index.js and time_source.js. For those 2 files the tsc compiler is complaining about their import of clock.js. Investigating and will share more asap.

wayneparrott, 2019-11-21:
Please disregard my previous note re: using the typescript compiler to generate ts declaration files from jsdoc annotated js files. After some experimentation I find the generated declaration files are missing a lot of information that I would expect as a typescript developer. That is, a lot "any" typed parameters, types and property values. Given this I will stick with my initial proposal to manually create and edit ts declaration files for rclnodejs.

minggangw, 2019-11-24:
Hi @wayneparrott , thank you for synchronizing the latest information of your current effort, I appreciate that. I think the 3 steps you mentioned at https://github.com/RobotWebTools/rclnodejs/issues/537#issuecomment-557164193 is reasonable, we could create a feature branch and merge it back to the `develop` finally.

I just saw another stable release of ROS2, [ROS 2 Eloquent Elusor](https://github.com/ros2/ros2/releases/tag/release-eloquent-20191122), has just been published yesterday, I hope we can enable our first TS edition `rclnodejs`  on it. 😄 By your valuable contribution, I believe this JS&TS client will attract more developers in the future.

I am not good at TypeScript, to be honest, please follow your plan to make it happen. But I can still do something like resolving the errors on CI or incompatibility caused by the `rcl` library which `rclnodejs` wraps. I'd like to help anything I could, thanks!

wayneparrott, 2019-11-25:
Let's precede with creating a feature branch that I'll work in. Today I upgrade my ros2 dev environment to eloquent. I'm assuming that you'll need to create the branch.

In a couple of days I will check in an initial PR consisting of a readme, declaration files, tests based on dtslint and a generator script similar to the rclnodejs generator but it will create a declaration file for all of the generated/ messages and services. TS declarations for messages is helpful with code completion and validation when coding a complex message that has a many primitive and complex fields.

minggangw, 2019-11-26:
I have created [typescript](https://github.com/RobotWebTools/rclnodejs/tree/typescript) branch, you can push your commits to it. Maybe you also have to add the branch name into the CI configuration files, e.g. [.travis.yml](https://github.com/RobotWebTools/rclnodejs/blob/develop/.travis.yml#L6), if you want to trigger it.

Currently, I don't create a branch again a specific ROS2 release, because there are many patch releases for a single version. But the `rclnodejs` may not have so much change at the meantime. So, usually I will run all unit tests with the latest `develop` or tag to verify if it can work well.

wayneparrott, 2019-11-27:
Thanks! I picked up the branch yesterday and have started making progress. Will share details asap. Taking a couple of holidays here in the US atm.

closed 2019-12-13 by wayneparrott