RobotWebTools/rclnodejs #541: Does the develop branch support ros2 eloquent?

wayneparrott, 2019-11-29:
Does the rclnodejs develop branch support the recent ros2 eloquent release?

I have an issue using eloquent with rclnodejs and wanted to check for known issues before starting to chase down the issue.

I updated to the recent ros2 eloquent release on my linux system as well as my rclnodejs develop branch. I'm experiencing a seg fault when I run script/run_tests.js as well as with a new ros2 workspace/pkg.  Previously on ros2 dashing I was able to run the rclnodejs tests successfully on my system. 

Lastly from a shell I can successfully run the ros2 cli with examples such as the publish/subscribe below: 
> ros2 topic pub /test std_msgs/msg/String "data: hello world" 
> ros2 topic echo /test std_msgs/msg/String

minggangw, 2019-11-30:
All the unit tests can pass on eloquent, please see #538, would you please offer which specific test crashed or which function caused the segment fault?

wayneparrott, 2019-11-30:
Disregard; my BAD! Working as expected.
I had goofed up my installation. 
Apologies for wasting your time.

closed 2019-11-30 by wayneparrott