RobotWebTools/rclnodejs #550: constants support in messages

koonpeng, 2020-01-18:
is there a way to reference the constant value like in rclpy? Something like
```
const marker = {...
  type: ImageMarker.CIRCLE,
...}
```
I can see that constants are generated in the js messages like so
```
// Define constants (7 in total)
Object.defineProperty(ImageMarkerWrapper, "CIRCLE", {value: 0, writable: false, enumerable: true, configurable: true});
Object.defineProperty(ImageMarkerWrapper, "LINE_STRIP", {value: 1, writable: false, enumerable: true, configurable: true});
Object.defineProperty(ImageMarkerWrapper, "LINE_LIST", {value: 2, writable: false, enumerable: true, configurable: true});
Object.defineProperty(ImageMarkerWrapper, "POLYGON", {value: 3, writable: false, enumerable: true, configurable: true});
Object.defineProperty(ImageMarkerWrapper, "POINTS", {value: 4, writable: false, enumerable: true, configurable: true});
Object.defineProperty(ImageMarkerWrapper, "ADD", {value: 0, writable: false, enumerable: true, configurable: true});
Object.defineProperty(ImageMarkerWrapper, "REMOVE", {value: 1, writable: false, enumerable: true, configurable: true});
```
But it needs to be imported like so
```
const ImageMarker = require('rclnodejs/generated/visualization_msgs/visualization_msgs__msg__ImageMarker');
```
which doesn't seem the intended usage.

Also, the typescript interfaces are generated like this
```
      export type ImageMarker = {
        CIRCLE: 0,
        LINE_STRIP: 1,
        LINE_LIST: 2,
        POLYGON: 3,
        POINTS: 4,
        ADD: 0,
        REMOVE: 1,
        header: std_msgs.msg.Header,
        ns: string,
        id: number,
        type: number,
        action: number,
        position: geometry_msgs.msg.Point,
        scale: number,
        outline_color: std_msgs.msg.ColorRGBA,
        filled: undefined,
        fill_color: std_msgs.msg.ColorRGBA,
        lifetime: builtin_interfaces.msg.Duration,
        points: geometry_msgs.msg.Point[],
        outline_colors: std_msgs.msg.ColorRGBA[]
      };
```
The constants have to be defined in objects that implements the type.
```
const marler: ImageMarker = {
  // All the constants have to be defined
  CIRCLE: 0,
  LINE_STRIP: 1,
...
};
```

minggangw, 2020-01-19:
Thanks for your question, I checked the usage of const variable defined in `.msg` files, and I think the code probably looks like:

```JavaScript
const ImageMarker = rclnodejs.require('visualization_msgs/msg/ImageMarker');
const LINE_LIST = ImageMarker.LINE_LIST;  // you get 2 here.
```
Not sure it's what you want or you could suggest a better usage, thanks!

minggangw, 2020-01-19:
Maybe I should introduce a little bit of the message objects used in `rclnodejs` to make it clear:

- You can use so called `plain` objects to pass a message to publisher/client, e.g.
```JavaScript
 const node = rclnodejs.createNode('publisher_message_example_node');
 const publisher = node.createPublisher('sensor_msgs/msg/JointState', 'JointState');
 publisher.publish({
      header: {
        stamp: {
          sec: 123456,
          nanosec: 789,
        },
        frame_id: 'main frame',
      },
      name: ['Tom', 'Jerry'],
      position: [1, 2],
      velocity: [2, 3],
      effort: [4, 5, 6],
    });
``` 
see https://github.com/RobotWebTools/rclnodejs/blob/develop/example/publisher-message-example.js

- You can also require a JS object which is defined by the generated classes (usually under `/generated`), e.g.
```JavaScript
let JointState = rclnodejs.require('sensor_msgs/msg/JointState');  // rclnodejs.require('sensor_msgs').msg.JointState;
let state = new JointState();
state.position = [1, 2];
state.velocity = [2, 3];
state.effort = [4, 5, 6];
```

Currently, most the usage is shown by the first way, because it's easier and more obvious and you can use `console.log(state)` to get the detailed values of it directly.

koonpeng, 2020-01-19:
Thanks! I wasn't aware of the `require` function. For the typescript types, would it be good to make the constants optionals? So there is no need to define them in an object, I am also looking at the possibly of generating the `require` type declaration so it returns a strongly typed interface instead of a generic `object`.

minggangw, 2020-01-19:
@wayneparrott added the TS support recently, I know he is busy implementing another important feature of rclnodejs, so Wayne would you please have a look if you have time? or @koonpeng you could help to submit a PR based on your requirement, thanks a lot!

koonpeng, 2020-01-19:
I have been using a tool I made to generate typescript interfaces, now that rclnodejs has support for typescript, it would be better to use the native support. I will try to see if I can make a PR for this, thanks!

wayneparrott, 2020-01-21:
@koonpeng  thanks for identifying my goof up on the type-alias msg constants. After reviewing, I'm not sure what I was thinking at the time I impl'ed this. I was experimented with a couple of approaches. Please share suggestions/PR for improvement. In the meantime I plan to carve out some time to look into resolving this asap as well.

wayneparrott, 2020-01-21:
A quick thought is to represent all constants for a type_class/msg as either an enum (idl constants are limited to primitives at this time) or an object literal similar to this:
```
export const ImageMarkerConstants = {
        CIRCLE: 0,
        LINE_STRIP: 1,
        LINE_LIST: 2,
        POLYGON: 3,
        POINTS: 4,
        ADD: 0,
        REMOVE: 1
}
export type ImageMarker = {
        header: std_msgs.msg.Header,
        ns: string,
        id: number,
        type: number,
        action: number,
...
```

Other ideas?

koonpeng, 2020-01-22:
normally this wouldnt work because `ImageMarkerConstants` doesn't exist in the javascript code. But I just found out there is a special case in 'const enum' where typescript will inline the value in the generated js code. I will see if I can add it to my PR.

koonpeng, 2020-01-22:
https://github.com/RobotWebTools/rclnodejs/pull/551 This PR improve typing information and generate constants as `const enum`

commit a04c6d22 (2020-02-17): better typescript typings (#551)

This commit makes the changes below:

* generate type class strings for messages and services
* narrow down `TypeClass` type from `string` to known values
* generate typings for *Wrapper javascript classes
* generate constants as 'const enum'

These provide intellisense for typeclass names and allow constants to be referenced by value.

The constants are generated with '_Constants' suffix, since ros does not allow underscores in
the message name, this should be safe from any conflicts. This is also same as how the ros idl
generator exports the constants.

Fix  [... cut]
closed 2020-02-17 by minggangw