RobotWebTools/rclnodejs #553: NodeJS version

lukaszmitka, 2020-01-21:
I was trying to use rclnodejs with NodeJS v8.17 and it ended up in segmentation fault when running examples. Upgrading to v10.18 solved the issue.
I am working on Ubuntu 18.04 and ROS Dashing Patch 4.
README states that NodeJS versions between 6.10 - 10.99 are compatible, did I do something wrong, or  README should be updated?

minggangw, 2020-01-21:
The minimum version of NodeJS, which is v6.10 on Ubuntu 16.04 LTS, was once verified on a very old release of ROS2 (maybe on the first stable release about 2 years ago). After that, we haven't verified it again, so it may become imprecise.

Would you please help to paste the core dump? If it's caused by the incompatibility of Node.js, we are going to update the README accordingly, thanks!

lukaszmitka, 2020-01-22:
I have run the example subscriber with valgrind: `valgrind -v --track-origins=yes node example/subscription-example.js`
The output is: https://rclnodejs-segfault.s3.eu-central-1.amazonaws.com/valgrind.log

And the core dump from running `node example/subscription-example.js`: https://rclnodejs-segfault.s3.eu-central-1.amazonaws.com/core_dump

minggangw, 2020-01-22:
Thanks, I am going to reproduce it locally and find any useful information, then get back.

minggangw, 2020-01-27:
I did some investigation locally and found that the crash happens when calling rcl_init_options_init() from librcl.so. It seems that it's not a problem caused by Node.js, I will dig deeper.

minggangw, 2022-06-20:
The supported nodejs versions are
https://github.com/RobotWebTools/rclnodejs/blob/f50e2b895c0a3129d8a85c6066cb17a031f94349/package.json#L92
Close this issue, please re-open it if you still have questions, thanks!

closed 2022-06-20 by minggangw