RobotWebTools/rclnodejs #587: how to get multiple topic data at the same time

EliotCao, 2020-02-28:
I am new about ros and I want get multiple topic data using Rclnodejs as description in subscription example, is there possibility?
appriciate for you help.

wayneparrott, 2020-02-28:
I believe what you are seeking is a [ros message_filter](https://github.com/ros2/message_filters) which to my understanding has not yet been implemented by this project.  Until it's implemented a fallback approach would be to implement a simple version that meets your project needs.

EliotCao, 2020-03-02:
thanks for your reply

closed 2020-03-02 by EliotCao

wayneparrott, 2020-04-10:
@EliotCao I've started looking into your question a bit deeper and while message_filters are how you can accomplish this in python or C, there is not yet a primary message_filter-like solution. If I had to solve this problem today I would give RxJS a hard look as it seems like it is made for these type of event and timing based problems. I'll follow up with an example soon if you're still looking at this problem.

EliotCao, 2020-04-21:
I'm not working on it, but it maybe valuable for others. thanks a lot.