RobotWebTools/rclnodejs #612: Tests fail when using the Eclipse cyclon dds

wayneparrott, 2020-03-18:
Environment:
- Ubuntu 18.04
- Eclipse Cyclone DDS 0.1.0:  installed from debian package ros-eloquent-rmw-cyclonedds-cpp
- export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

Background: 
https://index.ros.org/doc/ros2/Tutorials/Working-with-Eclipse-CycloneDDS/

Following is the console output from `npm run test` 
```
Cross-language interaction
    Node.js Subcription
      ✓ Node.js subscription should receive msg from C++ publisher (2987ms)
      ✓ Node.js subscription should receive msg from Python publisher (340ms)
    Node.js publisher
      ✓ Cpp subscription should receive msg from Node.js publisher (131ms)
node: /tmp/binarydeb/ros-eloquent-cyclonedds-0.1.0/src/core/ddsi/include/dds/ddsi/q_thread.h:190: 
thread_state_awake_fixed_domain: Assertion `ts1->state == THREAD_STATE_ALIVE' failed.
Aborted (core dumped)
```

minggangw, 2020-03-19:
It seems that the internal state of the thread is incorrect

```c
DDS_EXPORT inline void thread_state_awake_fixed_domain (struct thread_state1 *ts1)
{
  /* fixed domain -> must be an internal thread */
  assert (ts1->state == THREAD_STATE_ALIVE);
  thread_state_awake_domain_ok (ts1);
}
```
https://github.com/eclipse-cyclonedds/cyclonedds/blob/master/src/core/ddsi/include/dds/ddsi/q_thread.h#L211

wayneparrott, 2020-11-19:
I've been running ros2-foxy with Eclipse cyclone DDS since June 2020 with no issues.
Given that the status of this issues should be WORKSFORME. 
I recommend we close it.

wayneparrott, 2021-06-17:
With the ros2 galactic release (May '21) Eclipse Cyclone is now the default RMW. rclnodejs test suite successfully passes on the galactic distro. Thus, I am closing this issue as WORK FOR ME now. 

For galactic rmw info see https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#default-rmw-changed-to-eclipse-cyclone-dds

closed 2021-06-17 by wayneparrott