RobotWebTools/rclnodejs #631: How to call a service?

zentala, 2020-03-26:
I found how to create a service, but is there any way to call a service?

minggangw, 2020-03-26:
I suggest you find the examples [here](https://github.com/RobotWebTools/rclnodejs/tree/develop/example) and the API [doc](http://robotwebtools.org/rclnodejs/docs/index.html), that's what you need to know how to write a basic ROS2 app by rclnodejs.

zentala, 2020-03-26:
Thanks and sorry. I didn't notice this example as I have been looking for something with `service` in a name. It may be a good idea to rename this file to `service-client-exmaple.js`.

minggangw, 2020-03-30:
Thanks for your suggestion, so I think we could close it now. Please feel free to reopen it if you find anything wrong.

closed 2020-03-30 by minggangw