RobotWebTools/rclnodejs #640: Unable to install on Ubuntu 18.04LTS with Dashing ROS2 release

muawijhe, 2020-04-03:
Hi guys and girls,

`In file included from ../src/rcl_bindings.cpp:38:0:
../src/rcl_bindings.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE rclnodejs::GetParameterOverrides(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rcl_bindings.cpp:128:19: error: ‘rcl_arguments_get_param_overrides’ was not declared in this scope
       RCL_RET_OK, rcl_arguments_get_param_overrides(parsed_args, &params),
                   ^
`

npm -v
6.13.4

nodejs -v
v12.16.1

dpkg -l | grep ros-dashing-rclcpp | awk '{print $2 "\t" $3}'
ros-dashing-rclcpp	0.7.13-1bionic.20200318.040150
ros-dashing-rclcpp-action	0.7.13-1bionic.20200318.041055
ros-dashing-rclcpp-components	0.7.13-1bionic.20200318.041322
ros-dashing-rclcpp-lifecycle	0.7.13-1bionic.20200318.040923

what additional information do you need to help me out?

V.

wayneparrott, 2020-04-03:
@muawijhe thx for reporting this issue. I was able to replicate the issue using the dashing distro as well. The parameter feature that is failing to compile is fairly new and was tested in our CI system against the ros2 eloquent distro. An upgrade to the eloquent distro would be your quickest resolution. We need to investigate and see what is required for supporting previous ros2 version(s).

muawijhe, 2020-04-04:
Thanks for the reply and the quick response.

I believe it is important to maintain high and effective compatibility with
LTS ROS2 releases as Dashing.

People are relying on the LTS releases to create and maintain products.

Do you think that your interest project will soon be (again) usable for the
LTS release?

V.

muawijhe, 2020-04-04:
I also tried to use different versions without success. Did you try yourself with older version as well? If so coud you be so gentle and post here the version (and or the command)?

Thanks

minggangw, 2020-04-04:
Thanks for @wayneparrott's explanation and I want to share my thoughts about this problem.

- Which version should I use?

Usually, each release of `rclnodejs` targets on a (patch) release of ROS2. There are **7** releases of Dashing totally, including 1 main release and 6 patch release (the latest version is [patch release 6](https://github.com/ros2/ros2/releases/tag/release-dashing-20200319)). Meanwhile, we have 4 releases of rclnodejs accordingly, please see below: 

|                                             ROS 2.0 release                                             |                                                             NPM version                                                             |
| :-----------------------------------------------------------------------------------------------------: | :---------------------------------------------------------------------------------------------------------------------------------: |
| Dashing Diademata Patch Release [4](https://github.com/ros2/ros2/releases/tag/release-dashing-20191018) |                              [0.10.3](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.10.3)                               |
| Dashing Diademata Patch Release [3](https://github.com/ros2/ros2/releases/tag/release-dashing-20190910) |                              [0.10.2](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.10.2)                               |
| Dashing Diademata Patch Release [2](https://github.com/ros2/ros2/releases/tag/release-dashing-20190806) |                              [0.10.1](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.10.1)                               |
| Dashing Diademata Patch Release [1](https://github.com/ros2/ros2/releases/tag/release-dashing-20190614) |                                     [0.10.0](https://www.npmjs.com/package/rclnodejs/v/0.10.0)                                      |

We cannot ensure that rclnodejs runs correctly on patch relase 5&6, maybe  you could run ```npm i rclnodejs@0.10.3``` to have a try on these two releases:

But if you installed Dashing by [Debian packages](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/), you may not be aware of the exact release (I reckon it is patch release 6).

- About ROS2 LTS distro

Personally speaking, I don't think there is a kind of LTS distro until now, you could check out all the releases [here](https://index.ros.org/doc/ros2/Releases/#releases), although EOL of Dashing is May 2021. The community publishes a new release every 6 months followed by several patch releases. That will be changed after [Foxy](https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/).

Currently, the `develop` branch of rclnodejs tracks to `master` of ROS2, so we can keep it compatible as soon as a stable release of ROS2 is published. As @wayneparrott said, you are encouraged to install Eloquent at the current stage, unless you have strong reasons to stay on a specific ROS2 release.

muawijhe, 2020-04-04:
Dear Wang, thanks for your reply,

I believe Ubuntu is delivering patch 6 (based on what I see from ROS2
website).

I'll try this 0.10.3 and i'll give you feedback.
I understand that ROS2 has a release every 6 months, but again, people try
to stick to the long EOL release (unfortunately I named LTS), to manage
migration and support of products.
I hope you can help with this issue as Dashing will be supported longer
then successor and your project could be interesting for exposing ROS data
over the web.

Let me know if I might help someone to reduce bugfix time although I am
nothing good at javascript eheheg

V.

muawijhe, 2020-04-11:
Hi guys,
There is no one out there able to adjust the release for the current Ubuntu patch 6 version?

It will be great to support long term distro :(

minggangw, 2020-04-12:
We try to develop the rclnodejs client based on the latest stable ROS2, because we have limited resources to maintain several branches simultaneously (most of the engineers spend their free time in developing this project). Maybe we could support more after Foxy release as it will change less.

muawijhe, 2020-04-12:
`In file included from ../src/rcl_bindings.cpp:38:0: ../src/rcl_bindings.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE rclnodejs::GetParameterOverrides(Nan::NAN_METHOD_ARGS_TYPE)’: ../src/rcl_bindings.cpp:128:19: error: ‘rcl_arguments_get_param_overrides’ was not declared in this scope RCL_RET_OK, rcl_arguments_get_param_overrides(parsed_args, &params), ^

npm -v
6.13.4

nodejs -v
v12.16.1

dpkg -l | grep ros-dashing-rclcpp | awk '{print $2 "\t" $3}'
ros-dashing-rclcpp 0.7.13-1bionic.20200318.040150
ros-dashing-rclcpp-action 0.7.13-1bionic.20200318.041055
ros-dashing-rclcpp-components 0.7.13-1bionic.20200318.041322
ros-dashing-rclcpp-lifecycle 0.7.13-1bionic.20200318.040923`

This is the error. Tnx for letting me hope :)

closed 2025-02-19 by minggangw