RobotWebTools/rclnodejs #646: Support to publish/subscribe serialized messages

minggangw, 2020-04-10:
rclnodejs should support the capability of publishing/subscribing serialized messages, which offers by rcl library:

- [rcl_publish_serialized_message](https://github.com/ros2/rcl/blob/7537f8304493e9b9568748b5601f5c1b7fa463f0/rcl/include/rcl/publisher.h#L359)
- [rcl_take_serialized_message](https://github.com/ros2/rcl/blob/7537f8304493e9b9568748b5601f5c1b7fa463f0/rcl/include/rcl/subscription.h#L308)

minggangw, 2020-04-12:
Anybody who is interested in this issue,  please take it freely.

wayneparrott, 2020-07-04:
I was curious about how serialized messages might be implemented for subscriptions and publishers. I took a look at [rclpy's implementation of rcl_take for subscriptions](https://github.com/ros2/rclpy/blob/87390606b218169ab07ca2f99c7c6480d6f8eed5/rclpy/src/rclpy/_rclpy.c) which uses a parameter to specify when to use call `rcl_take_serialized_message`.

With that as background here's some quick thoughts. I'm assuming the rclnodejs Subscription constructor would include an option for when to return a serialized message (i.e., uint8 buffer). Additionally rclnodejs.Message would be extended to return a serialized message.

Example:
```
  Node.createSubscription(
     typeclass: rclnodejs.MessageClass, 
     topic: String, 
     options: Options, 
     callback: (rclnodejs.Message)=>void
  )

 interface Options<T = QoS | QoS.ProfileRef> {
    enableRaw?: boolean = false;
    enableTypedArray?: boolean = true;
    qos?: T;
  }
```

Thoughts?

minggangw, 2020-07-05:
> I'm assuming the rclnodejs Subscription constructor would include an option for when to return a serialized message (i.e., uint8 buffer).

+1 we could simply add a key like `enableRaw`  to take this parameter instead of a separate one like `rclpy` implementation.

> Additionally rclnodejs.Message would be extended to return a serialized message.

We already have it, https://github.com/RobotWebTools/rclnodejs/blob/03e84dc5264a5f4363fb20c89cadd3f392cfceee/lib/publisher.js#L54-L55
`rawMessage ` is an instance of `Buffer`, so I think we can leverage the node [Buffer](https://nodejs.org/api/buffer.html) to pass the serialized messages.

rclpy's PR for the serialized publisher: https://github.com/ros2/rclpy/pull/509
rclpy's PR for the serialized subscription: https://github.com/ros2/rclpy/pull/242
node Buffer header file: https://github.com/nodejs/node/blob/master/src/node_buffer.h

minggangw, 2020-09-30:
Reopen it because the subscription part hasn't been done.

PR #703 (2020-09-30): Support to publish serialized messages
This patch implements the feature that a publisher can publish a raw
message, and an example is added.

Fix #646
commit f574a343 (2020-09-30): Support to publish serialized messages

This patch implements the feature that a publisher can publish a raw
message, and an example is added.

Fix #646
PR #704 (2020-10-02): Support to subscribe serialized messages
This patch implements the feature that a subscription can subscribe a raw
message. Meanwhile, an example showing the usage is added and a unit
test for the serialized publisher & subscription.

Fix #646
commit 25282cef (2020-10-02): Support to subscribe serialized messages

This patch implements the feature that a subscription can subscribe a raw
message. Meanwhile, an example showing the usage is added and a unit
test for the serialized publisher & subscription.

Fix #646
closed 2020-10-02 by minggangw