RobotWebTools/rclnodejs #657: Support waitsets

wayneparrott, 2020-05-23:
rclcpp recently added a new waitset api, https://github.com/ros2/rclcpp/issues/520 Is the waitset capability something we want to consider for inclusion in rclnodejs?

minggangw, 2020-05-24:
Thanks for the information! It's a big refactor inside the rclcpp framework and I go through the motivation of it roughly (I see many issues were connected to this PR), the most important one I think is to reduce the CPU overhead. But the changes are based on the design of rclcpp specifically and we have some limitations of threads in Node.js, so I think we need to investigate more to see how will it benefit to developers if we expose kind of `waitset` in JavaScript and the influence to the current framework of rclnodejs.

minggangw, 2020-05-25:
Just some thoughts in case I forget them:

- From the current framework of rclnodejs, we don't have kind of `Executor` (SingleThreadedExecutor/MultiThreadedExecutor) exposed to the JavaScript layer, the concept of the thread is transparent to the developers. Further, the `Executor` in rclnodejs runs on the sub-thread because it's not allowed to block the main thread in Node.js

- Exposing `waitset` means users can call `wait()` to block the runloop, some cpp examples about the new usage https://github.com/ros2/examples/pull/262/files

wayneparrott, 2020-05-25:
>I think we need to investigate more to see how will it benefit to developers if we expose kind of waitset in JavaScript and the influence to the current framework of rclnodejs.

My blink reaction was similar to your's.  Let's revisit as ros 2 continues to mature and additional features emerge that we want incorporated into the project.