RobotWebTools/rclnodejs #658: generated msg declarations in typescript do not include typed-arrays

wayneparrott, 2020-05-28:
This is a general question about how typescript declarations for ros msgs represent numeric array properties. For example the `data` property of a ros2 image is defined as  uint8[] in [IDL](https://github.com/ros2/common_interfaces/blob/master/sensor_msgs/msg/Image.msg). Yet the typescript interface generated by rclnodejs (see below) declares the data property as a `number[]`. 
```
export interface Image {
        header: std_msgs.msg.Header;
        height: number;
        width: number;
        encoding: string;
        is_bigendian: number;
        step: number;
        data: number[];
 }
```
Shouldn't `data` in the interface be defined as `number[] | Uint8Array` ?

minggangw, 2020-05-29:
> Shouldn't data in the interface be defined as number[] | Uint8Array ?

I agree as rclnodejs can adopt both types.

wayneparrott, 2020-07-04:
fixed.

closed 2020-07-04 by wayneparrott