RobotWebTools/rclnodejs #659: When will the Foxy Fitzroy version be supported?

jacobbubu, 2020-06-06:
I am new to this project, maybe someone has already answered this question before, but I did not find.

Thanks!

wayneparrott, 2020-06-06:
Setting up an env to test drive foxy tonight. Will provide feedback asap.

wayneparrott, 2020-06-06:
Good news! I was able to successfully run the full project test suite. Given that I'll be switching to use the ROS2 Foxy release.

minggangw, 2020-06-06:
> Good news! I was able to successfully run the full project test suite. Given that I'll be switching to use the ROS2 Foxy release.

Really good news :)

@jacobbubu  Fox was just released on June 5th, so I am going to verify rclnodejs with it on all supported platforms. Hopefully, we are going to publish a new version with fixing some minor bugs, e.g. #649, thanks!

jacobbubu, 2020-06-07:
I try to run test cases of rclnodejs under macOS Catalina. I tested Node.JS v14.3.0 and Node.JS v12.13.1 respectively, and encountered different problems (I will reinstall `rm -rf node_modules && npm i && npm run install` before execution).

Under NodeJS v12, I encountered 8 test cases that failed, as shown below:
[Failure cases on Node.JS v12](https://www.dropbox.com/s/hru3qklqjc051ae/ros2-foxy-nodejs12.png?dl=0)

In NodeJS v14, the use case cannot be executed at all:
[Failure cases on Node.JS v14](https://www.dropbox.com/s/vper6q5leh0cdfq/ros2-foxy-nodejs14.png?dl=0)

BTW, I'm working on the develop branch https://github.com/RobotWebTools/rclnodejs/commit/3e93c48cb32d3d7a8f09fce59676782f9d79fb7e.

minggangw, 2020-06-07:
Thanks for helping to test the cases on different Node.js versions, some thoughts:

> Under NodeJS v12, I encountered 8 test cases that failed, as shown below:
Failure cases on Node.JS v12

Those are known failures because some cases used to test interaction with rclpy/rclcpp tend to be flaky. Also, I noticed that you opened #661 which reports an error while installing `rclnodejs` on Node.js v12, so I am wondering if you meet the error actually? (you could reference the #660 for the detailed configuration)

> In NodeJS v14, the use case cannot be executed at all:
Failure cases on Node.JS v14

`rclnodejs` has not been compiled on Node.js v14 (tested with v12 currently), so there may be some issues.

jacobbubu, 2020-06-07:
@minggangw The #661 error is caused by installing directly from npm, which is different from #659 trying to build from source

minggangw, 2020-06-07:
Oh, I see and sorry for misunderstanding your issue. The failure is because the rclnodejs v0.14.0 is incompatible with ROS2 Foxy release, while the `develop` branch has land the patch which fixed that. So, I suggest you could wait for the next release to be officially published and if you want to run your code with Foxy right now, you could install from `develop` branch instead. Thanks!

wayneparrott, 2020-06-07:
@jacobbubu 
> I try to run test cases of rclnodejs under macOS Catalina.
Apologies when I incompletely reported the test suite was running on foxy. I should have included I was testing on a linux 20.04 VM. I'm on a mac as well that I use for a wide range of dev.  The prereq's, etc discouraged me from installing it directly. Call me a scaredy cat :)

minggangw, 2020-06-10:
[v0.14.1](https://www.npmjs.com/package/rclnodejs/v/0.14.1) is released, please upgrade your rclnodejs to the latest version, enjoy!

minggangw, 2020-06-10:
I suppose we could close it now, please feel free to reopen it if you still have problems, thanks!

closed 2020-06-10 by minggangw