RobotWebTools/rclnodejs #661: Fatal error encountered while installing rclnodejs on macOS Catalina & Node.JS v12

jacobbubu, 2020-06-07:
```
 rongshen@rs   ~/tmp/ros2/pub  npm i rclnodejs -S                                                                                    10099  11:44:07 

> int64-napi@1.0.1 install /Users/rongshen/tmp/ros2/pub/node_modules/int64-napi
> node-gyp rebuild

  CC(target) Release/obj.target/int64/src/int64.o
../src/int64.c:55:54: warning: incompatible pointer types passing 'uint64_t *' (aka 'unsigned long long *') to parameter of type 'size_t *'
      (aka 'unsigned long *') [-Wincompatible-pointer-types]
        napi_get_buffer_info(env, _raw, (void**)raw, &rawLength);
                                                     ^~~~~~~~~~
/Users/rongshen/Library/Caches/node-gyp/12.13.1/include/node/node_api.h:142:54: note: passing argument to parameter 'length' here
                                             size_t* length);
                                                     ^
1 warning generated.
  SOLINK_MODULE(target) Release/int64.node

> ref-napi@2.0.2 install /Users/rongshen/tmp/ros2/pub/node_modules/ref-napi
> node-gyp-build

> rclnodejs@0.14.0 install /Users/rongshen/tmp/ros2/pub/node_modules/rclnodejs
> node-gyp rebuild

  CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
  CXX(target) Release/obj.target/rclnodejs/src/handle_manager.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_action_bindings.o
../src/rcl_action_bindings.cpp:631:5: warning: ignoring return value of function declared with 'warn_unused_result' attribute [-Wunused-result]
    rcl_action_cancel_response_fini(cancel_response_ptr);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~~~~~~~~~~~~
1 warning generated.
  CXX(target) Release/obj.target/rclnodejs/src/rcl_bindings.o
../src/rcl_bindings.cpp:32:10: fatal error: 'rosidl_generator_c/string_functions.h' file not found
#include <rosidl_generator_c/string_functions.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [Release/obj.target/rclnodejs/src/rcl_bindings.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/rongshen/.nvm/versions/node/v12.13.1/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Darwin 19.5.0
gyp ERR! command "/Users/rongshen/.nvm/versions/node/v12.13.1/bin/node" "/Users/rongshen/.nvm/versions/node/v12.13.1/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/rongshen/tmp/ros2/pub/node_modules/rclnodejs
gyp ERR! node -v v12.13.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN pub@1.0.0 No description
npm WARN pub@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rclnodejs@0.14.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the rclnodejs@0.14.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/rongshen/.npm/_logs/2020-06-07T03_44_41_932Z-debug.log
```

SIP has already disabled.

minggangw, 2020-06-07:
Thanks for your question, and the failure is because the rclnodejs v0.14.0 is incompatible with ROS2 Foxy release.

jacobbubu, 2020-06-08:
@minggangw I tested under Node.JS v12.13.1.

minggangw, 2020-06-08:
Thanks for the information, the compiling error is due to the change brought by rcl library instead of Node.js versions related problems.

minggangw, 2020-06-10:
[v0.14.1](https://www.npmjs.com/package/rclnodejs/v/0.14.1) is released, please upgrade your rclnodejs to the latest version, enjoy!

jacobbubu, 2020-06-11:
Confirm that it can be successfully installed under Node.JS v12 and macOS Catalina .But under Node.JS v14 I will still encounter segmentation fault.

closed 2020-06-11 by jacobbubu

minggangw, 2020-06-11:
@jacobbubu would you please open a new issue to track the current problem (segmentation fault) on Node.js v14 and offer the backtrace of it which will help us to fix it or know what happens there, thanks!

zentala, 2020-06-18:
I have the same error: 
```
../src/rcl_action_bindings.cpp:631:36: warning: ignoring return value of ‘rcl_ret_t rcl_action_cancel_response_fini(rcl_action_cancel_response_t*)’, declared with attribute warn_unused_result [-Wunused-result]
     rcl_action_cancel_response_fini(cancel_response_ptr);
     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
  CXX(target) Release/obj.target/rclnodejs/src/rcl_bindings.o
../src/rcl_bindings.cpp:33:10: fatal error: rosidl_runtime_c/string_functions.h: No such file or directory
 #include <rosidl_runtime_c/string_functions.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

...under:
* Ubuntu 18.04
* ROS2 Eloqunet
* Node v12.16.2

...trying to install package from NPM or build it manually

wayneparrott, 2020-06-18:
@zentala 
Sorry you are hitting this issue. Try installing as follows:
```
npm i rclnodejs@0.14.0
```
The most recent rclnodejs version is 0.14.1 which is only compatible with the latest ros2 foxy release. The rclnodejs v0.14.0 is the most recent version compatible with ros2 eloquent. 

Action item will be to revisit the docs to see if we can make the version compatibility info more prominent on npmjs.

minggangw, 2020-06-18:
> Action item will be to revisit the docs to see if we can make the version compatibility info more prominent on npmjs.

I strongly agree as users tend to install the latest version by default, and the [table](https://github.com/RobotWebTools/rclnodejs#match-with-ros-20-stable-releases) we offered was ignored usually. Unlike rclpy/rclcpp, rclnodejs is not integrated with ROS2 package, so users have to select a proper version to match with their ROS2 (considering there are already many ROS2 release & patch release 😄)

jacobbubu, 2020-06-19:
> @jacobbubu would you please open a new issue to track the current problem (segmentation fault) on Node.js v14 and offer the backtrace of it which will help us to fix it or know what happens there, thanks!

There is no problem installing rclnodejs@0.14.1 under Node.JS v14.3.0, but there will be some compilation warnings.

```
> int64-napi@1.0.1 install /Users/rongshen/trash/rcln-test/node_modules/int64-napi
> node-gyp rebuild

  CC(target) Release/obj.target/int64/src/int64.o
../src/int64.c:55:54: warning: incompatible pointer types passing 'uint64_t *'
      (aka 'unsigned long long *') to parameter of type 'size_t *'
      (aka 'unsigned long *') [-Wincompatible-pointer-types]
        napi_get_buffer_info(env, _raw, (void**)raw, &rawLength);
                                                     ^~~~~~~~~~
/Users/rongshen/Library/Caches/node-gyp/14.3.0/include/node/node_api.h:144:54: note:
      passing argument to parameter 'length' here
                                             size_t* length);
                                                     ^
1 warning generated.
  SOLINK_MODULE(target) Release/int64.node

> ref-napi@2.0.2 install /Users/rongshen/trash/rcln-test/node_modules/ref-napi
> node-gyp-build

> rclnodejs@0.14.1 install /Users/rongshen/trash/rcln-test/node_modules/rclnodejs
> node-gyp rebuild

  CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
  CXX(target) Release/obj.target/rclnodejs/src/handle_manager.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_action_bindings.o
../src/rcl_action_bindings.cpp:631:5: warning: ignoring return value of function declared with 'warn_unused_result' attribute [-Wunused-result]
    rcl_action_cancel_response_fini(cancel_response_ptr);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~~~~~~~~~~~~
1 warning generated.
  CXX(target) Release/obj.target/rclnodejs/src/rcl_bindings.o
../src/rcl_bindings.cpp:1315:24: warning: 'New' is deprecated: Use the version that takes a BackingStore. See http://crbug.com/v8/9908.
      [-Wdeprecated-declarations]
      v8::ArrayBuffer::New(v8::Isolate::GetCurrent(), address, length,
                       ^
/Users/rongshen/Library/Caches/node-gyp/14.3.0/include/node/v8.h:5103:3: note: 'New' has been explicitly marked deprecated here
  V8_DEPRECATE_SOON(
  ^
/Users/rongshen/Library/Caches/node-gyp/14.3.0/include/node/v8config.h:402:39: note: expanded from macro 'V8_DEPRECATE_SOON'
# define V8_DEPRECATE_SOON(message) [[deprecated(message)]]
                                      ^
1 warning generated.
  CXX(target) Release/obj.target/rclnodejs/src/rcl_handle.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_utilities.o
  CXX(target) Release/obj.target/rclnodejs/src/shadow_node.o
  SOLINK_MODULE(target) Release/rclnodejs.node

> rclnodejs@0.14.1 postinstall /Users/rongshen/trash/rcln-test/node_modules/rclnodejs
> node scripts/generate_messages.js

Start JavaScript message generation...
Generation complete.
npm WARN notsup Unsupported engine for rclnodejs@0.14.1: wanted: {"node":">= 8.12.0 <13.0.0"} (current: {"node":"14.3.0","npm":"6.14.5"})
npm WARN notsup Not compatible with your version of node/npm: rclnodejs@0.14.1
npm WARN rcln-test@1.0.0 No description
npm WARN rcln-test@1.0.0 No repository field.

+ rclnodejs@0.14.1
added 44 packages from 75 contributors and audited 44 packages in 29.833s
found 0 vulnerabilities
```

minggangw, 2020-06-19:
Got it, thanks @jacobbubu