RobotWebTools/rclnodejs #673: Possible use after free in callbacks

koonpeng, 2020-07-14:
I created a new branch `bug/callback-uaf` to help reproduce it. It adds a `test-promise-gc` test case that fails with what looks like the result a use-after-free.

From my experiments, the uaf occurs when we assign the argument of of a callback that contains a typed array to a variable from the outer scope

```js
      const result = await new Promise((res) =>
        node.createSubscription(typeClass, topic, (msg) => {
          timer.cancel();
          node.destroy();
          res(msg);
        }),
      );
      global.gc();
```

here, `msg` is assigned to `result` through the promise resolver, after which we manually trigger the gc. The typed array ref in `result` now points to invalid memory.

Some points to note:
  * Without the gc call, the test passes but it may already refer to invalid memory.
  * Running in valgrind doesn't seem to catch anything.
  * In electron, this causes the array to contain all `0`s, even without the manual gc call, perhaps electron has different gc behavior.

Some initial debugging:

In `node.js`:

```js
  _runWithMessageType(Type, callback) {
    let message = new Type();

    callback(message.toRawROS(), () => {
      let result = new Type();
      result.deserialize(message.refObject);

      return result;
    });

    Type.destoryRawROS(message);
  }
```

`destroyRawROS` is called after the callback is ran, this eventually call the native function `CreateArrayBufferCleaner`. It creates a new `RclHandle`, which implements some RAII functions that `free` the underlying address when it goes out of scope. This `RclHandle` is wrapped in a `v8::Local<v8::Object>` and under a `EscapeHandleScope`.

  * Is there possibility for a `RclHandle` to get copied, and end up with multiple instance of `RclHandle` holding the same pointer, resulting in double frees when they go out of scope?
  * The return value of `CreateArrayBufferCleaner` is not assigned to anything, would it cause the underlying `RclHandle` to go out of scope immediately, freeing the address?
  * Where does the address that `RclHandle` is holding comes from? Is there any chance we are freeing addresses that we are not managing?

minggangw, 2020-07-15:
Thanks for investigating this issue, I think I can explain a little bit of the background of the usage of the typed array.

1. During the deserialization

https://github.com/RobotWebTools/rclnodejs/blob/03e84dc5264a5f4363fb20c89cadd3f392cfceee/rosidl_gen/templates/message.dot#L729-L736

When the typed array is enabled (|usePlainTypedArray| equals to true), we are going to call a  binding method to create an array buffer from a specific address, see

https://github.com/RobotWebTools/rclnodejs/blob/87375dedd4faaaa77376118d076864bb3d03111e/src/rcl_bindings.cpp#L1321-L1330

The memory block starting from |address| is allocated typically by the rcl api, e.g. `rcl_take`, so here the typed array is actually backed by the memory allocated by rcl library. The reason is that we don't have to copy the data which saves resource and time. The relationship looks like:

typed arrary => v8 array buffer => memory allocated by `rcl`

2. After the deserialization

https://github.com/RobotWebTools/rclnodejs/blob/95729f9a3650547b372fa1dc2c37d3d7f1d85d82/rosidl_gen/templates/message.dot#L444-L446

The memory allocated by `rcl` is deallocated immediately, so the typed array becomes invalid.

3.  Considerations

Users may transfer huge data by typed array e.g. `uint8`,  if we depend on GC to reclaim the memory, we have to copy the original memory block and the memory consumption could be very high as when the memory is GCed is unpredictable. We ensure the typed array is valid in one loop and expect users to copy it himself on demand.

 So I think 

> The return value of CreateArrayBufferCleaner is not assigned to anything, would it cause the underlying RclHandle to go out of scope immediately, freeing the address?

is correct. Do you have any suggestion we could improve here, shall we have an extra memcpy?

koonpeng, 2020-07-16:
Thanks for the insight, I am still new to node native modules and v8, I'm trying to see if there is a way to let v8 manage the lifecycle of the memory allocated by `rcl_take` so that it is only free-ed when there is no more reference in javascript.

minggangw, 2020-07-16:
I landed a patch bbcc0503ade29f09936c896b7cf211a5bb818bb0 to fix the double free problem and I can pass the test locally, but I'm not sure there is no memory leak by using this, @koonpeng would you please take a look?

minggangw, 2020-07-17:
https://github.com/RobotWebTools/rclnodejs/blob/87375dedd4faaaa77376118d076864bb3d03111e/src/rcl_bindings.cpp#L1325-L1327

> Create a new ArrayBuffer over an existing memory block. The created array buffer is by default immediately in externalized state. The memory block will not be reclaimed when a created ArrayBuffer is garbage-collected.

That's the v8 doc description about [```ArrayBuffer::New```](https://v8docs.nodesource.com/node-7.10/d5/d6e/classv8_1_1_array_buffer.html#a98c19b0bf91cc3ea22371fbab5707e25). So I think we still have to take care of it because it will not get GCed.

With patch bbcc050, run ***valgrind*** with command 
```bash
valgrind --tool=memcheck --leak-check=full node --expose-gc node_modules/.bin/mocha test/test-promise-gc.js -f 'std_msgs/msg/UInt8MultiArray'
``` 
reports:

```bash
==1249961== 10 bytes in 1 blocks are definitely lost in loss record 9 of 104
==1249961==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==1249961==    by 0x927D94F: rosidl_runtime_c__uint8__Sequence__init (primitives_sequence_functions.c:69)
==1249961==    by 0xFDF03AF: _UInt8MultiArray__cdr_deserialize (u_int8_multi_array__type_support_c.cpp:121)
==1249961==    by 0x95C4AD2: rmw_fastrtps_cpp::TypeSupport::deserializeROSmessage(eprosima::fastcdr::Cdr&, void*, void const*) const (type_support_common.cpp:97)
==1249961==    by 0x9619F95: rmw_fastrtps_shared_cpp::TypeSupport::deserialize(eprosima::fastrtps::rtps::SerializedPayload_t*, void*) (TypeSupport_impl.cpp:100)
==1249961==    by 0x993D4C0: eprosima::fastrtps::SubscriberHistory::deserialize_change(eprosima::fastrtps::rtps::CacheChange_t*, unsigned int, void*, eprosima::fastrtps::SampleInfo_t*) (SubscriberHistory.cpp:307)
==1249961==    by 0x9941EE7: eprosima::fastrtps::SubscriberHistory::takeNextData(void*, eprosima::fastrtps::SampleInfo_t*, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >&) (SubscriberHistory.cpp:384)
==1249961==    by 0x9936BBF: eprosima::fastrtps::SubscriberImpl::takeNextData(void*, eprosima::fastrtps::SampleInfo_t*) (SubscriberImpl.cpp:124)
==1249961==    by 0x9617D89: _take (rmw_take.cpp:77)
==1249961==    by 0x9617D89: rmw_fastrtps_shared_cpp::_take(char const*, rmw_subscription_t const*, void*, bool*, rmw_message_info_t*, rmw_subscription_allocation_t*) (rmw_take.cpp:52)
==1249961==    by 0x924296A: rcl_take (subscription.c:269)
==1249961==    by 0x92078AA: rclnodejs::RclTake(Nan::FunctionCallbackInfo<v8::Value> const&) (in /home/minggang/proj/ros2/rclnodejs-1/build/Release/rclnodejs.node)
==1249961==    by 0x91F665B: Nan::imp::FunctionCallbackWrapper(v8::FunctionCallbackInfo<v8::Value> const&) (in /home/minggang/proj/ros2/rclnodejs-1/build/Release/rclnodejs.node)

```

The result shows the block was leaked indeed.

minggangw, 2020-07-17:
I updates the patch 31fb53fa87f6f9d996630355548ae9152d2f3155. `test-promise-gc.js` can pass now and there is no memory leak of the typed array from valgrind. But the downside is that we have to sacrifice the performance, @koonpeng what do you think? Is it suitable? If it's acceptable, we could have a hotfix.

koonpeng, 2020-07-18:
https://v8docs.nodesource.com/node-12.16/d5/d6e/classv8_1_1_array_buffer.html#a98c19b0bf91cc3ea22371fbab5707e25 The docs here says if we set the array buffer to `kInternalized` v8 will free the address when there are no more reference. Maybe that would work to automatically clean up the buffer?

minggangw, 2020-07-18:
Oh, I even didn't notice that... and have changed to pass `kInternalized`. The unit tests and valgrind don't complain anymore, perfect 😄 So, I submitted #679 to be reviewed. Would you please submit another PR to include `test-promise-gc.js`  (please remove any copy right about Intel brand, you could use Apache License or your company's if you like), thanks!

commit bbcc0503 (2020-07-16): Fix double-free when using typed array

This patch enables to transfer the ownership of ArrayBuffer to v8
when deserializing the messages, which has type of TypedArray.
This will avoid double-free when the v8 executes gc.

Fix #673
commit 31fb53fa (2020-07-17): Fix double-free when using typed array

This patch enables to transfer the ownership of ArrayBuffer to v8
when deserializing the messages, which has type of TypedArray.
This will avoid double-free when the v8 executes gc.

Fix #673
commit 354ed208 (2020-07-18): Fix double-free when using typed array

This patch enables to transfer the ownership of ArrayBuffer to v8
when deserializing the messages, which has type of TypedArray.
This will avoid double-free when the v8 executes gc.

Fix #673
PR #679 (2020-07-20): Fix double-free when using typed array
This patch enables to transfer the ownership of ArrayBuffer to v8
when deserializing the messages, which has type of TypedArray.
This will avoid double-free when the v8 executes gc.

Fix #673
closed 2020-07-20 by minggangw