RobotWebTools/rclnodejs #687: No TypeScript support for ActionClient destroy() method

caelinsutch, 2020-07-30:
If you look at the [API documentation](http://robotwebtools.org/rclnodejs/docs/0.15.0/lib_action_client.js.html), a destroy method is documented. In the [type declaration file](https://github.com/RobotWebTools/rclnodejs/blob/develop/types/action_client.d.ts) no such method is documented. 

I'm opening to submitting a PR for this issue.

mattrichard, 2020-07-31:
Thanks for reporting this. The type declaration was definitely missing for the `destroy` function. Although, I ended up creating a PR before reading your last line. Feel free to submit PRs for any issues you encounter in the future.

closed 2020-07-31 by minggangw

caelinsutch, 2020-08-01:
When's the estimated release date for v0.16.0?

minggangw, 2020-08-03:
> When's the estimated release date for v0.16.0?

Currently, we don't have a specific milestone of rclnodejs, but there are some cases that we'll have a feature/bugfix release:

- A new release of [ROS2](https://github.com/ros2/ros2/releases), we always follow the latest ROS2 tightly and verify rclnodejs on it.
- New features are added, e.g. TypeScript support, Action support.
- Critical bugs are fixed, e.g. memory leak...

So, do you want we could publish a bugfix version ASAP to fix this issue?

caelinsutch, 2020-08-03:
> So, do you want we could publish a bugfix version ASAP to fix this issue?

If you could do a bugfix with this fix and the PR for #688 that would be awesome.

minggangw, 2020-08-04:
Of course, we could do this. Hopefully, I am going to release 0.15.2 this week, stay tuned.

minggangw, 2020-08-07:
@caelinsutch please check out v0.15.2 https://www.npmjs.com/package/rclnodejs/v/0.15.2

PR #690 (2020-07-31): Add missing type declarations for action server/client destroy methods
Action servers and clients are not created through a node and therefore don't have corresponding destroy methods on the Node class. Instead, to destroy an action server/client outside of destroying the Node itself it is expected to call `destroy` on the server/client directly. This adds the type declarations for the `destroy` function which were missing.

Fix #687