RobotWebTools/rclnodejs #688: Types for creating an Action Client not Updating with Sources Action Message Types

caelinsutch, 2020-07-30:
Problem example:

```js
// Variable actionType has to be a certain string value. Types don't update when sourcing additional packages
const newActionClient = new ActionClient(node, actionType as ActionTypeClassName, actionName);

newActionClient.sendGoal(data).then(v => {
  log.debug('Action Complete!');
})
```

mattrichard, 2020-07-31:
Right now the message types are only generated during postinstall or at runtime when calling `init()`. I can see how this would be a problem with TypeScript since you need to be able to compile everything before `init()` can be run again. We'll either need to make the action types more flexible or find better ways to update the generated message typings when additional packages are sourced.

For now, you can workaround this issue by manually regenerating the message typings using the following command. This should at least unblock you until there's a proper solution.

```bash
node node_modules/rclnodejs/scripts/generate_messages.js
```

minggangw, 2020-07-31:
We once planed to implement that the message generator could generate the JS messages from IDL dynamically if the required message cannot be found, but can be found through `PATH` (the new package will be added after you source the `xxx_install.sh`). 

Unfortunately, you have to manually delete the `generated\` folder in order to trigger to generate the messages completely again, or workaround as @mattrichard said. Thanks!

caelinsutch, 2020-08-01:
> For now, you can workaround this issue by manually regenerating the message typings using the following command. This should at least unblock you until there's a proper solution.

I got this error:
```
Start JavaScript message generation...
Caught error: TypeError: Cannot read property 'fields' of undefined
```

caelinsutch, 2020-08-01:
Update - Even after deleting `generated` and even the entire folder and reinstalling, I'm still not getting all my packages sourced. For example, when calling a the service `/save_maps` from Slamtoolbox, I'm getting the error `The message required does not exist: slam_toolbox, srv, SaveMap`. In the same terminal window, I can run that service from the command line. 

This is from the CLI: 

```

➜ ros2 service type /save_map 
slam_toolbox/srv/SaveMap

```

wayneparrott, 2020-08-02:
Just saw your updates. Thx for sharing.
I installed slam_toolbox and believe the error message you shared is from the generate_tsd.js script resulting in the interfaces.d.ts file having incomplete slam_toolbox message types defined. Will dig into this asap.
```
Start JavaScript message generation...
Caught error: TypeError: Cannot read property 'fields' of undefined
```
Can you confirm in the node_modules/rclnodejs/generated folder that the following files are created:
```
slam_toolbox__srvs__SaveMap.js
slam_toolbox__srvs__SaveMap_Request.js
slam_toolbox__srvs__SaveMap_Response.js
``` 
Here's what my generated/slam_toolbox and my types/interfaces.d.ts look like: 
[image]

wayneparrott, 2020-08-03:
I'm able to isolate the problem to the slam_toolbox does not follow the ROS naming convention for placing services interfaces (idl) in a folder named `srv`. Instead slam_toolbox services idl lives in a folder named `srvs` [here](https://github.com/SteveMacenski/slam_toolbox/tree/foxy-devel/srvs).

I'll submit a PR with that is a bit more robust handling of such situations. 

A short term workaround is to manually change 1 line of code.
In node_modules/rclnodejs/rclgen_tsd/index.js line as shown below.
[image]

PR #691 (2020-08-04): Improve tsd generation resilency.
rostsd_gen/index.js
- support service idl defined in any folder starting with 'srv', e.g.,
'srvs'
-  support action idl defined in any folder starting with 'action',
e.g., 'actions'

Fix #688
commit 9523d212 (2020-08-04): Improve tsd generation resilency.

rostsd_gen/index.js
- support service idl defined in any folder starting with 'srv', e.g.,
'srvs'
-  support action idl defined in any folder starting with 'action',
e.g., 'actions'

Fix #688
commit 4b4761b1 (2020-08-07): Improve tsd generation resilency.

rostsd_gen/index.js
- support service idl defined in any folder starting with 'srv', e.g.,
'srvs'
-  support action idl defined in any folder starting with 'action',
e.g., 'actions'

Fix #688
closed 2020-08-04 by minggangw