RobotWebTools/rclnodejs #689: See When Action Completes

caelinsutch, 2020-07-31:
I can't figure out how to implement a callback on an `ActionClient` when the action completes, just on updates and when the action is successfully received. As mentioned in line 190 [here](http://robotwebtools.org/rclnodejs/docs/0.15.0/lib_action_client.js.html):

```
  /**
   * Send a goal and asynchronously get the result.
   *
   * The result of the returned Promise is set to a ClientGoalHandle when receipt of the goal
   * is acknowledged by an action server.
   *
   * @param {object} goal - The goal request.
   * @param {function} feedbackCallback - Callback function for feedback associated with the goal.
   * @param {object} goalUuid - Universally unique identifier for the goal. If None, then a random UUID is generated.
   * @returns {Promise} - A Promise to a goal handle that resolves when the goal request has been accepted or rejected.
   */
```

However, on line 313, I see the function `_getResult`:

```
  /**
   * Request the result for an active goal asynchronously.
   * @ignore
   * @param {ClientGoalHandle} goalHandle - Handle to the goal to cancel.
   * @returns {Promise} - A Promise that resolves when the get result request has been processed.
   */
```

Is this something that should be implemented, or an accepted restriction?

mattrichard, 2020-07-31:
The comments do appear somewhat misleading. The `sendGoal` function returns a Promise which resolves to a goal handle. To get the final result (i.e. the completion) you need to call `getResult()` on the goal handle and wait for that Promise to resolve. You can see this in the examples here

https://github.com/RobotWebTools/rclnodejs/blob/db7721ec3042b355d45089cb77aec9c34c6577b8/example/action-client-example.js#L52

That promise will resolve when the action completes, and it will contain the final result and status. Let me know if you need more info.

minggangw, 2020-07-31:
Shall we update the comments accordingly to make it clear?

caelinsutch, 2020-08-01:
I think the docs should be updated for sure.

PR #692 (2020-08-07): Correct the description of sendGoal function
Fix #689
commit 53685f78 (2020-08-07): Correct the description of sendGoal function

Fix #689
commit 5715fc3c (2020-08-07): Correct the description of sendGoal function

Fix #689
closed 2020-08-07 by minggangw