RobotWebTools/rclnodejs #694: Getting a Fatal Error when Working with Large Messages

caelinsutch, 2020-08-10:
After calling a service to get two large map arrays, I get the following fatal error on some (low compute) machines. Any advice on how to get past this? 

```
#
# Fatal error in , line 0
# Check failed: length == backing_store->byte_length().
#
#
#
#FailureMessage Object: 0x7fcd01b8d0
 1: 0xa57754  [node]
 2: 0x185cb20 V8_Fatal(char const*, ...) [node]
 3: 0xdf85e8 v8::internal::GlobalBackingStoreRegistry::Lookup(void*, unsigned long) [node]
 4: 0xb787d0  [node]
 5: 0xb789e4 v8::ArrayBuffer::New(v8::Isolate*, void*, unsigned long, v8::ArrayBufferCreationMode) [node]
 6: 0x7fa4e95194 rclnodejs::CreateArrayBufferFromAddress(Nan::FunctionCallbackInfo<v8::Value> const&) [/home/bytes/Github/satellitum/packages/robot/node_modules/rclnodejs/build/Release/rclnodejs.node]
 7: 0x7fa4e883fc  [/home/bytes/Github/satellitum/packages/robot/node_modules/rclnodejs/build/Release/rclnodejs.node]
 8: 0xbb7534  [node]
 9: 0xbb85f8  [node]
10: 0xbb8c78 v8::internal::Builtin_HandleApiCall(int, unsigned long*, v8::internal::Isolate*) [node]
11: 0x1329f4c  [node]
[1]    42276 trace trap (core dumped)  ts-node robot.ts
```

minggangw, 2020-08-10:
> I've found this issue that looks like it might relate? I'm guessing the buffer is too small for the message, but that doesn't make sense considering it works fine on other machines...

It seems that we meet the same case here, using an external buffer to create an ArrayBuffer, is it reproducible stably?

caelinsutch, 2020-08-10:
> It seems that we meet the same case here, using an external buffer to create an ArrayBuffer, is it reproducible stably?

Yes, running on a Jetson Nano this is reproducible. Whenever any service is called that sends a map array it spits out that error :(

caelinsutch, 2020-08-10:
I'm now getting this error:
```
# Fatal error in , line 0
# Check failed: result.second.
```

Found [this discussion](https://github.com/nodejs/node/issues/32463) on buffers

minggangw, 2020-08-11:
It seems that this problem is not simple to be resolved and I haven't dug into it, maybe it's a hardware related issue? Anyway,  we could disable the usage of TypedArray during message serialization, e.g.

https://github.com/RobotWebTools/rclnodejs/blob/b71c8eab85b23b50132bf4d78f2f16bb2b1a1cbf/test/test-disable-typedarray.js#L60-L70

The downside is that you lose the performance. please try it to see if it works.

caelinsutch, 2020-08-11:
I was using Node 14, when I switched back to 12 the issue resolved itself, so this isn't necessarily a bug in the supported versions of RCLNodeJS and I've closed the issue.

closed 2020-08-11 by caelinsutch

minggangw, 2020-08-12:
good to hear you have a workaround.