RobotWebTools/rclnodejs #698: Using rclnodejs in a react-app doesn't work

MaximeAubanel, 2020-09-02:
Hello,

System: docker
Base Image: Ubuntu 20.04
ROS: Foxy
Node: 12.x LTS

I'm trying to create a ROS2 node using rclnodejs with Typescript inside a create-react-app but I'm having a warning when I launch the app.

```
./node_modules/rclnodejs/lib/interface_loader.js
Critical dependency: the request of a dependency is an expression
```

Do you have any ideas about what could cause this ?

MaximeAubanel, 2020-09-02:
From what I understood, I cannot run rclnodejs on the client side because `rclnodejs` needs `fs` can you guys confirm this ? Thanks :)

minggangw, 2020-09-02:
> CRA doesn't provide

What's CRA?  Is it create-react-app?

The rclnodejs does depend on `fs` module and you could reference https://github.com/RobotWebTools/rclnodejs/blob/foxy-fitzroy/Dockerfile about the usage of docker.

MaximeAubanel, 2020-09-02:
Yes sorry, CRA is create-react-app.

When I was importing `import * as rclnodejs from 'rclnodejs'` as follow in my typescript app, I was having this error `TypeError: exists is not a function`.  Leading me to `fs`. Maybe one of rclnodejs dependencies use `fs` ?

minggangw, 2020-09-02:
https://github.com/RobotWebTools/rclnodejs/blob/95729f9a3650547b372fa1dc2c37d3d7f1d85d82/rosidl_gen/index.js#L42-L43

rclnodejs uses it directly when generating JavaScript messages. If you have installed it, there should be no error reported I think.

MaximeAubanel, 2020-09-02:
AFAIK react app does not allow the use of `fs`.

See https://github.com/facebook/create-react-app/issues/3074#issuecomment-327484250

That would explain it. 
So it gives me no choice to have it on the backend side, which is indeed the best solution but I just wanted to do a quick Proof Of Concept.

minggangw, 2020-09-03:
I see, so I think we could close this issue now. Please feel free to add any comments if you find a workaround.

closed 2020-09-03 by minggangw

flynneva, 2020-09-03:
@MaximeAubanel I was working on a wrapper of the roslibjs library for react awhile ago. havent made much progress on it since COVID but its still there if you'd like to check it out. [react-ros on npm](https://www.npmjs.com/package/react-ros) and then the source code should be linked in there as well.