RobotWebTools/rclnodejs #705: Cannot be used with Electron on Windows

fstojanovic, 2020-09-30:
It would appear that it is not possible to run with Electron on Windows.

When run if built for node, the following message pops up : 

> App threw an error during load
> Error: The module '\\?\C:\Users\{SNIP}\rclnodejs\build\Release\rclnodejs.node'
> was compiled against a different Node.js version using
> NODE_MODULE_VERSION 72. This version of Node.js requires
> NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
> the module (for instance, using `npm rebuild` or `npm install`).
>     at process.func (electron/js2c/asar.js:138:31)
>     at process.func [as dlopen] (electron/js2c/asar.js:138:31)
>     at Object.Module._extensions..node (internal/modules/cjs/loader.js:828:18)
>     at Object.func (electron/js2c/asar.js:138:31)
>     at Object.func [as .node] (electron/js2c/asar.js:138:31)
>     at Module.load (internal/modules/cjs/loader.js:645:32)
>     at Function.Module._load (internal/modules/cjs/loader.js:560:12)
>     at Module.require (internal/modules/cjs/loader.js:685:19)
>     at require (internal/modules/cjs/helpers.js:16:16)
>     at bindings (C:\Users\{SNIP}\rclnodejs\node_modules\bindings\bindings.js:112:48)
> 

When I try to rebuild it for Electron using electron-rebuild, the build always fails, due to missing headers and dependencies. Once I manually resolve all those errors and it finally builds, running it for electron gives : "rclnodejs.node" is not a module.

When run for Node, without Electron, it works as expected.

minggangw, 2020-10-02:
I haven't used rclnodejs with Electron, I think @koonpeng has some experiences with Electron env, @koonpeng would you please have a look at this issue? Have you ever met sort of the error? Some comments, hope it helpful

> When run if built for node, the following message pops up :

I don't think it could be workable with nodejs build, as you saw `was compiled against a different Node.js version using`
was reported.

> due to missing headers and dependencies. 

What's the header missed? Is it ROS2 headers or what else?

> running it for electron gives : "rclnodejs.node" is not a module.

Can you find the `rclnodejs.node` under the `build/` folder? It seems that the lib doesn't get generated.

koonpeng, 2020-10-02:
There is a bug in the current version of electron which causes rclnodejs to not work, the fix is currently scheduled in v11 beta. Also, if you are loading rclnodejs in the renderer process you need to disable renderer reuse (there's a flag somewhere in electron api). This is because rclnodejs is not "context-aware" and reusing the same process can cause unexpected behaviours.

That being said, the error "rclnodejs.node is not a module" may be caused by some problems in the compilation process. You can start by checking if the file exists and see if there is any error messages during the compilation.

minggangw, 2020-10-02:
Thanks for @koonpeng 's quick explanation, that's helpful!

>  This is because rclnodejs is not "context-aware"

I think we could improve this?

fstojanovic, 2020-10-04:
@minggangw , @koonpeng 

Thank you for your replies. I will keep you posted about the issue, soon I'll update this issue thread.
Also, it doesn't build for any Electron version, not just the current one.

newcanopies, 2021-01-21:
@fstojanovic does `rclnodej`s build with `electron` on WSL2 Ubuntu 20.04?

Thanks all!

adeelarshad83, 2022-02-09:
> @minggangw , @koonpeng
> 
> Thank you for your replies. I will keep you posted about the issue, soon I'll update this issue thread. Also, it doesn't build for any Electron version, not just the current one.

Hi @fstojanovic, are you able to resolve this issue

minggangw, 2024-09-26:
@fstojanovic please check with rclnodejs Electron demo just added https://github.com/RobotWebTools/rclnodejs/tree/develop/electron_demo

closed 2024-09-26 by minggangw