RobotWebTools/rclnodejs #711: Problem with space in path

felixdivo, 2020-10-15:
It seems like a space in the path causes the Javascript messages file building to crash:

```
> rclnodejs@0.15.1 postinstall /shortened-path-for-privacy/some space/nodejs-project/node_modules/rclnodejs
> node scripts/generate_messages.js

Start JavaScript message generation...
Caught error: Error: python3: can't open file '/shortened-path-for-privacy/some': [Errno 2] No such file or directory
```

Software versions:
- ROS2 Foxy
- Ubuntu 20.04
- Node.js v14.10.1
- rclnodejs 0.15.1

I couldn't yet test whether moving the project solves the issue.

minggangw, 2020-10-16:
Thanks for submitting this issue, we are going to fix it.

felixdivo, 2020-10-16:
Thanks. Without the space, I merely get a segfault right after that step. I'll report details once I diagnosed it further.

minggangw, 2020-10-16:
#699 has fixed the segfault, please try the latest version.

felixdivo, 2020-10-16:
Thank you, that indeed solved the issue!

PR #712 (2020-10-20): Escape spaces in path
This patch implements to escape space in path before passing it to
python.

Fix #711
commit 52de37bc (2020-10-20): Escape spaces in path

This patch implements to escape space in path before passing it to
python.

Fix #711
commit 87cad092 (2020-11-26): Escape spaces in path

This patch implements to escape space in path before passing it to
python.

Fix #711
closed 2020-10-20 by minggangw