RobotWebTools/rclnodejs #713: segfault with node v12.19.0

koonpeng, 2020-10-20:
rclnodejs segfaults after a few seconds after receiving a subscription. It only happens with node v12.19, works fine in v12.18.4.

To reproduce:

```js
const rclnodejs = require('rclnodejs');

async function main() {
  await rclnodejs.init();
  const node = rclnodejs.createNode('test');
  rclnodejs.spin(node);
  
  node.createSubscription('std_msgs/msg/String', 'test', {}, console.log);

  const pub = node.createPublisher('std_msgs/msg/String', 'test');
  pub.publish({ data: 'hello' });
}

main();
```

Output:
```bash
$ node test.js 
{ data: 'hello' }
Segmentation fault (core dumped)
```

minggangw, 2020-10-20:
I noticed the same issue when I tried to upgrade the Node.js to v12.19 for the CI, and after checking the dump file, the backtrace shows no rclnodejs related callings. I suggest we could verify this on the next stable release of Node.js, how do you think?

minggangw, 2020-10-20:
I cannot reproduce it on v13.14.0

minggangw, 2020-10-29:
I try the latest Node.js LTS v14.15.0 #716 and it seems that there is a crash caused by `ref-napi`, paste below:

```bash
#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 0x7fff04fd3a70
 1: 0xa70141  [node]
 2: 0x19cf084 V8_Fatal(char const*, ...) [node]
 3: 0xe594c9 v8::internal::GlobalBackingStoreRegistry::Register(std::shared_ptr<v8::internal::BackingStore>) [node]
 4: 0xba4a18 v8::ArrayBuffer::GetBackingStore() [node]
 5: 0x9c18f0 napi_get_typedarray_info [node]
 6: 0x7fa130ce70ef  [/root/rclnodejs/node_modules/ref-napi/prebuilds/linux-x64/node.napi.node]
 7: 0x7fa130ce7918  [/root/rclnodejs/node_modules/ref-napi/prebuilds/linux-x64/node.napi.node]
 8: 0x7fa130ce7bbb  [/root/rclnodejs/node_modules/ref-napi/prebuilds/linux-x64/node.napi.node]
 9: 0x7fa130cefceb Napi::details::CallbackData<void (*)(Napi::CallbackInfo const&), void>::Wrapper(napi_env__*, napi_callback_info__*) [/root/rclnodejs/node_modules/ref-napi/prebuilds/linux-x64/node.napi.node]
```

koonpeng, 2020-10-29:
I got that as well, I think it's unrelated, it seems more like some v8 changes that causes it instead.

Btw, the offending commit that causes v12.19 to fail has been identified and I think they will be backporting the fix in the next release. I'll go ahead and close this issue.

closed 2020-10-29 by koonpeng

minggangw, 2020-11-17:
Segmentation fault still happens on node v12.19.1 when running `npm test`
```base
#0  0x00000000055001c0 in ?? ()
#1  0x0000000000d17c4b in v8::internal::GlobalHandles::InvokeSecondPassPhantomCallbacksFromTask() ()
#2  0x0000000000c953e3 in v8::internal::CancelableTask::Run() ()
#3  0x0000000000a86d54 in node::PerIsolatePlatformData::RunForegroundTask(std::unique_ptr<v8::Task, std::default_delete<v8::Task> >) ()
#4  0x0000000000a87a15 in node::PerIsolatePlatformData::FlushForegroundTasksInternal() ()
#5  0x000000000136f0ae in uv__async_io (loop=0x2cb3a80 <default_loop_struct>, w=<optimized out>, 
    events=<optimized out>) at ../deps/uv/src/unix/async.c:163
#6  0x0000000001382165 in uv__io_poll (loop=loop@entry=0x2cb3a80 <default_loop_struct>, 
    timeout=<optimized out>) at ../deps/uv/src/unix/linux-core.c:461
#7  0x000000000136f8ef in uv_run (loop=0x2cb3a80 <default_loop_struct>, mode=UV_RUN_DEFAULT)
    at ../deps/uv/src/unix/core.c:385
#8  0x0000000000a5aac6 in node::NodeMainInstance::Run() ()
#9  0x00000000009e85cc in node::Start(int, char**) ()
#10 0x00007f0ccfa3a0b3 in __libc_start_main (main=0x9804a0 <main>, argc=3, argv=0x7ffd273d8e98, 
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffd273d8e88)
    at ../csu/libc-start.c:308
#11 0x00000000009819b5 in _start ()

```

koonpeng, 2020-11-17:
:(
The fix seems to not be included, this is the line that cause the segfault. https://github.com/nodejs/node/blob/f95d7152cb9f56b3d6ca8e107a01c199b235c0df/src/js_native_api_v8.cc#L246.

It's been removed on master https://github.com/nodejs/node/blob/08be03b7cba11ea3aa17a9881b6ead3d8936abe3/src/js_native_api_v8.cc#L

minggangw, 2020-11-17:
Let's move to v14.x directly :sweat_smile:

koonpeng, 2020-11-17:
Actually v14 also segfaults

minggangw, 2020-11-17:
Yeah, indeed. It seems node.js is not stable enough :(

minggangw, 2020-11-25:
The segfaults has been fixed on the latest v12.20.0 :clap:

PR #740 (2020-11-25): Update Node.js to v12.20.0
Fix #713
commit 8821edf9 (2020-11-25): Update Node.js to v12.20.0

Fix #713
commit 7b19103a (2020-11-26): Update Node.js to v12.20.0

Fix #713