RobotWebTools/rclnodejs #728: Improve RclHandle ptr free() strategy

wayneparrott, 2020-11-19:
@koonpeng recently proposed redesigning the RclHandle to make the deleter lambda function responsible for `free()`ing the RclHandle memory pointer rather than `free()`ing its ptr directly as it does now. 

tldr;
The [RclHandle](https://github.com/RobotWebTools/rclnodejs/blob/develop/src/rcl_handle.cpp) `Reset()` method divides clean up duties between its self and a custom deleter lambda function provided during instance initialization. The deleter function performs custom delete activities on the RclHandle's ptr. While the RclHandle directly `free()`s its memory ptr. Recently this split in responsibilities resulted in a double `free()` error where the RclHandle ptr was free()'ed by a parent RclHandle and then again freed by its RclHandle wrapper.. The resolution required a creative hack using a custom deleter function that tricked the RclHandler to bypass calling free() on its ptr. This scenario revealed the need to externalize all of the clean up of the RclHandler memory ptr. The RclHandle deleter function is a natural candidate for the memory ptr free() responsibility.

minggangw, 2020-11-27:
Ping, anybody are looking at this issue, if not I am going to take it, thanks!

wayneparrott, 2020-11-28:
@minggangw pls proceed.

PR #743 (2020-12-02): Use specified deleter for RclHandle
This patch will enable RclHandle to use a specified deleter to free
the resources, which means the pointer passed when contructing the
RclHandle object will not be freed by default.

After this patch, caller should be responsible to manage the memory.

Fix #728
commit 04d9539b (2020-12-02): Use specified deleter for RclHandle

This patch will enable RclHandle to use a specified deleter to free
the resources, which means the pointer passed when contructing the
RclHandle object will not be freed by default.

After this patch, caller should be responsible to manage the memory.

Fix #728
commit 2b756425 (2021-01-08): Use specified deleter for RclHandle

This patch will enable RclHandle to use a specified deleter to free
the resources, which means the pointer passed when contructing the
RclHandle object will not be freed by default.

After this patch, caller should be responsible to manage the memory.

Fix #728
closed 2020-12-02 by minggangw