RobotWebTools/rclnodejs #729: Nav2_msg require error

adamkrawczyk, 2020-11-20:
version: rclnodejs@0.16.0 
ros: ros2 foxy

when:

`const nav2_msgs = rclnodejs.require('nav2_msgs').msg;`

what:

```
/home/adam/dev_ws/src/route_admin_panel/nodejs/node_modules/rclnodejs/lib/interface_loader.js:86
Object.defineProperty(pkg[info.type], info.name, {
              ^
 
TypeError: Object.defineProperty called on non-object
at Function.defineProperty (<anonymous>)
at /path/nodejs/node_modules/rclnodejs/lib/interface_loader.js:86:14
at Array.forEach (<anonymous>)
at Object.loadInterfaceByPath (/parth/nodejs/node_modules/rclnodejs/lib/interface_loader.js:85:20)
at Object.loadInterfaceByString (/path/nodejs/node_modules/rclnodejs/lib/interface_loader.js:65:19)
at Object.loadInterface (/path/nodejs/node_modules/rclnodejs/lib/interface_loader.js:100:21)
at Object.require (path/nodejs/node_modules/rclnodejs/index.js:334:19)
at Object.<anonymous> (/path/nodejs/main.js:20:29)
at Module._compile (internal/modules/cjs/loader.js:1015:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:1035:10)
```

Why: 

I have no idea why this occurs. Other messages are imported properly. Nav2_msgs are generated.

minggangw, 2020-11-20:
If `nav2_msgs` is not a ROS2 build-in package (released with ROS2 package), I think you have to source its `local_setup.bash` so the message generator could find this package and create the JS file for it.

wayneparrott, 2020-11-20:
nav2_msgs includes a number of actions. The error is due to `interface_loader.loadInterfaceByPath()` not being updated to load actions. 

The fix is to change line 84 from:
```
    let pkg = { srv: {}, msg: {} };
```
to
```
    let pkg = { srv: {}, msg: {}, action: {} };
```

minggangw, 2020-11-20:
Oh, I missed @adamkrawczyk has mentioned that 

> Nav2_msgs are generated.

So I think the `.bash` has been sourced. Thanks for @wayneparrott's findings.

adamkrawczyk, 2020-11-20:
Yeah! 

This is solving the issue. (`rclnodejs/lib/interface_loader.js   84: let pkg = { srv: {}, msg: {}, action: {} }; `)

When the npm version will be patched? 

Thank you ;)

closed 2020-11-20 by adamkrawczyk

wayneparrott, 2020-11-20:
@minggangw I can submit a PR with fix and test case if no one else is planning to work this issue. Thoughts?

minggangw, 2020-11-20:
@wayneparrott Please submit a PR to fix this issue, thanks! 

> When the npm version will be patched?

Hopefully, the next release v0.17.0 can land the patch and we will publish it once the current [issues](https://github.com/RobotWebTools/rclnodejs/projects/12) get addressed.

PR #730 (2020-11-21): Fix action support to interface_loader.loadInterfaceByPath()
**Commit Summary**

interface_loader.js
* fixed bug in loadInterfaceByPath() when loading action messages

test-message-object.js
* added 'require() and create instance of every message in package' unit
test

Fix #729
commit a5a84bda (2020-11-21): Fix action support to interface_loader.loadInterfaceByPath()

**Commit Summary**

interface_loader.js
* fixed bug in loadInterfaceByPath() when loading action messages

test-message-object.js
* added 'require() and create instance of every message in package' unit
test using tf2_msgs

Fix #729
commit 0a207146 (2020-11-26): Fix action support to interface_loader.loadInterfaceByPath()

**Commit Summary**

interface_loader.js
* fixed bug in loadInterfaceByPath() when loading action messages

test-message-object.js
* added 'require() and create instance of every message in package' unit
test using tf2_msgs

Fix #729
PR #793 (2021-05-27): support foxy & galactic simultaneously
ros_distro.js
- added ros galactic version id: 2105

rcl_lifecycle_binding.cpp
- uses ros galactic version id (2105) to select between foxy and
galactic lifecycle api

test-raw-pub-sub.js
- uses ros_distro.js to access ros version id
- for galactic version, test uses null-terminated string

test-node.js
- added small 100ms wait between publisher creation and test for
node.publisherCount() to avoid failing on slower systems.

fixes #729