RobotWebTools/rclnodejs #735: Compilation warning on newer node versions: v8 has a new ArrayBuffer BackingStore API

felixdivo, 2020-11-21:
I get this output by `node-gyp rebuild` (invoked using `scripts/build.sh`) on Node version `v14.10.1`:

```
../src/rcl_bindings.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE rclnodejs::CreateArrayBufferFromAddress(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rcl_bindings.cpp:1330:70: warning: ‘static v8::Local<v8::ArrayBuffer> v8::ArrayBuffer::New(v8::Isolate*, void*, size_t, v8::ArrayBufferCreationMode)’ is deprecated: Use the version that takes a BackingStore. See http://crbug.com/v8/9908. [-Wdeprecated-declarations]
 1330 |                            v8::ArrayBufferCreationMode::kInternalized);
      |                                                                      ^
In file included from /home/felix/.cache/node-gyp/14.10.1/include/node/node.h:67,
                 from ../node_modules/nan/nan.h:56,
                 from ../src/rcl_bindings.hpp:18,
                 from ../src/rcl_bindings.cpp:15:
/home/felix/.cache/node-gyp/14.10.1/include/node/v8.h:5169:29: note: declared here
 5169 |   static Local<ArrayBuffer> New(
      |                             ^~~
../src/rcl_bindings.cpp:1330:70: warning: ‘static v8::Local<v8::ArrayBuffer> v8::ArrayBuffer::New(v8::Isolate*, void*, size_t, v8::ArrayBufferCreationMode)’ is deprecated: Use the version that takes a BackingStore. See http://crbug.com/v8/9908. [-Wdeprecated-declarations]
 1330 |                            v8::ArrayBufferCreationMode::kInternalized);
      |                                                                      ^
In file included from /home/felix/.cache/node-gyp/14.10.1/include/node/node.h:67,
                 from ../node_modules/nan/nan.h:56,
                 from ../src/rcl_bindings.hpp:18,
                 from ../src/rcl_bindings.cpp:15:
/home/felix/.cache/node-gyp/14.10.1/include/node/v8.h:5169:29: note: declared here
 5169 |   static Local<ArrayBuffer> New(
      |                             ^~~```
```

I didn't include it in #734 as I don't really know any v8 internals.

minggangw, 2020-11-21:
#723 is supposed to solve the compiling error on Node.js >= 14

minggangw, 2020-11-21:
@felixdivo thanks for your contributions recently, we have a small group of teams where developers could plan/discuss for the rclnodejs project, and if you are interested in and want to contact others closely, I'd like to invite you to join the teams, thanks!

wayneparrott, 2020-11-21:
@felixdivo Join rclnodejs developers on Slack —  Sign up here, from any device: https://join.slack.com/t/rclnodejsdev/shared_invite/zt-jjukoa3l-zZ5VeAeo~YD11vmUC2Mdnw

minggangw, 2020-11-21:
Oops, sorry I mean slack, Teams is used in Intel...

felixdivo, 2020-11-21:
> #723 is supposed to solve the compiling error on Node.js >= 14

Ah great, so we can just add "closes # 635" to that PR. Nice that it's already being tackled!

I've kinda used up my current free time with a few contributions the last days :P but when I'm feeling like doing some more in this project, I'm happy to join! Thanks! :)

minggangw, 2020-11-22:
Never mind, none of us works on the project full-time, so welcome 🚀

minggangw, 2020-11-27:
Hi @felixdivo I haven't seen you in the slack channel of rclnodejs, if you met any problem, please send me your email address and I will send the invitation again.

closed 2022-01-03 by wayneparrott

felixdivo, 2022-01-04:
Hey @minggangw, sorry for not writing about this, but I retreated to other projects.

PR #825 (2022-01-03): Add node 13-17 support, bump to v0.21.0
**Public API Changes**
none

**Description**

0. package.json
* Bumped to version 0.21.0
* Support for node 13-16 required forking and customizing the following
packages:
@rclnodejs/ref-array-di
@rclnodejs/ref-napi
@rclnodejs/ref-struct-di
Revised package dependencies to use the 3 packaged listed above
* Added jsdoc as devDependency

2. test-security-related.js
* Fixed minor assertion error caused by change in Nodejs error message.

3. Docs
* Updated README.md, Build.md and npmjs-readme.md to expand valid
versions of node 13-16.
* Modified docs/Makefile  [... cut]