RobotWebTools/rclnodejs #765: Error in generated service request files

Jopand, 2021-01-05:
I'm running a Debian 10 environment with ROS2 Foxy and rclnodejs 16.
I tried to make a simple Angular page calling a change_state service from ROS (using ros2-web-bridge). It didn't work.

After some digging and debugging, I found out that it was due to an error in ./rclnodejs/generated/lifecycle_msgs/lifecycle_msgs__srv__ChangeState_Request.js

[image]

The generated line 31 is:
`translator.constructFromPlanObject(this, other);`
If I send other.request as the 2nd argument, it all works:
`translator.constructFromPlanObject(this, other.request);`

This is not just the case with the ChangeState service, but also some other homemade services. The above fix works with these services as well

minggangw, 2021-01-06:
`rclnodejs` acts as the back-end of `ros2-web-bridge` to connect to the ROS2, please try to run https://github.com/RobotWebTools/rclnodejs/tree/develop/example to check if you can reproduce the error using rclnodejs, thanks!

Jopand, 2021-01-08:
I just created a clean Ubuntu vm with ros2 foxy and couldn't reproduce the problem. So guess I have had a "dirty" setup before. Think we can close the issue

closed 2021-01-08 by Jopand