RobotWebTools/rclnodejs #778: Not able to run the module inside electron app

purvikpatel, 2021-03-12:
when I am trying to subscribe a message in an electron app it's giving me the following error.

Error: Loading non-context-aware native module in renderer: '/ROVDashboard/node_modules/rclnodejs/build/Release/rclnodejs.node', but app.allowRendererProcessReuse is true. 
`
can someone tell me how can I solve this error ?

minggangw, 2021-03-13:
What's your Electron version and platform? It was once reported there was a problem with Electron, please see #705

purvikpatel, 2021-03-15:
@minggangw I got the error similar to you but solved that by recompiling the modules. After that I got this error.

Can you tell me where you included your rclnodejs code in your electron project ?
main process or render process ?

minggangw, 2021-03-15:
I am not familiar with how the Electron loads the nodejs addon, AFAIK, there are two v8 isolates, one is running as the normal v8 instance in the render process, the other is running the nodejs (but not sure if it's in render process too,  but I guess it's running in the main process as the render process should be sandboxing).

purvikpatel, 2021-03-16:
The issue was with electron. As soon as I updated electron the issue was solved.

closed 2021-03-16 by purvikpatel