RobotWebTools/rclnodejs #784: wrapping rclnodejs client-side in Browser (chrome, ffx)

newcanopies, 2021-04-06:
Hi, is it possible to integrate rclnodejs in the Browser client-side HTML, so that it can be leveraged online without desktop apps?

minggangw, 2021-04-06:
Because rclnodejs needs the support of Node.js, if you want to integrate it into the browser, you have to support Node.js, e.g. Electron or NW.js, or you can use [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge) to run a ROS2 app in a common browser, e.g chrome

newcanopies, 2021-04-06:
>  or you can use [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge) to run a ROS2 app in a common browser, e.g chrome

thank you @minggangw , 
please clarify if by "ROS2 app" you mean to build an Electron or NW.js app containing ros2-web-bridge and my-ros-project? 

does that mean  [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge) would still need to be wrapped in Electron/NW.js?  (using either of these wrappers is not portable or cross platform among other downsides)

newcanopies, 2021-04-06:
Here is an example of javascript in HTML:

`
var message = {}
  var image = []
  rclnodejs.init().then(() => {
    const node = rclnodejs.createNode('subscription_example_node');
    
    node.createSubscription('sensor_msgs/msg/Imu', '/robot/imu', (msg) => {
    //console.log(`Received message: ${typeof msg['orientation']}`, msg['orientation']);
    message = msg
    
    win.webContents.send('imu', message)
      //console.log(count)
      //count += 1
    });
    node.createSubscription('sensor_msgs/msg/Image', '/robot/camera/image_raw', (m) => {
        //image = msg['data']
        console.log("msg received")
    })
    node.spin();
    });
`

newcanopies, 2021-04-10:
@minggangw can rclnodejs be wrapped in a react-app ?

minggangw, 2021-04-11:
Sorry for missing your previous comments, I will comment on them together

> does that mean ros2-web-bridge would still need to be wrapped in Electron/NW.js?

Actually, you have to depend on [roslibjs](https://github.com/RobotWebTools/roslibjs) (as front-end running in the browser) + [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge) (as back-end) and you will not need Electron any more, just Chrome/Edge will work well. Please check out some examples https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/examples/html

> can rclnodejs be wrapped in a react-app ?

If you are using roslibjs, of course, you can wrap it.

closed 2021-04-20 by newcanopies