RobotWebTools/rclnodejs #792: Compile error in version 0.19

ManuelZ, 2021-05-26:
Using Docker, I have Node 12.20.1 installed through NVM. I've installed ros2-web-bridge, which has rclnodejs 0.x.1 as a dependency. Just now I discovered this compile error, which wasn't really ocurring 5 days ago.

```
 > [build 15/15] RUN source /opt/ros/foxy/setup.bash     && npm install --global --unsafe-perm ros2-web-bridge:
#22 8.559 /root/.nvm/versions/node/v12.20.1/bin/rosbridge -> /root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/bin/rosbridge.js
#22 8.564
#22 8.564 > int64-napi@1.0.1 install /root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/node_modules/int64-napi
#22 8.564 > node-gyp rebuild
#22 8.564
#22 9.662 make: Entering directory '/root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/node_modules/int64-napi/build'
#22 9.664   CC(target) Release/obj.target/int64/src/int64.o
#22 9.926   SOLINK_MODULE(target) Release/obj.target/int64.node
#22 9.954   COPY Release/int64.node
#22 9.964 make: Leaving directory '/root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/node_modules/int64-napi/build'
#22 9.972
#22 9.972 > ref-napi@3.0.2 install /root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/node_modules/ref-napi
#22 9.972 > node-gyp-build
#22 9.972
#22 10.04
#22 10.04 > rclnodejs@0.19.0 install /root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/node_modules/rclnodejs
#22 10.04 > node-gyp rebuild
#22 10.04
#22 10.59 make: Entering directory '/root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/node_modules/rclnodejs/build'
#22 10.60   CXX(target) Release/obj.target/rclnodejs/src/addon.o
#22 11.23   CXX(target) Release/obj.target/rclnodejs/src/executor.o
#22 11.96   CXX(target) Release/obj.target/rclnodejs/src/handle_manager.o
#22 12.83   CXX(target) Release/obj.target/rclnodejs/src/rcl_action_bindings.o
#22 14.10   CXX(target) Release/obj.target/rclnodejs/src/rcl_bindings.o
#22 15.94   CXX(target) Release/obj.target/rclnodejs/src/rcl_handle.o
#22 16.69   CXX(target) Release/obj.target/rclnodejs/src/rcl_lifecycle_bindings.o
#22 17.12 ../src/rcl_lifecycle_bindings.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE rclnodejs::CreateLifecycleStateMachine(Nan::NAN_METHOD_ARGS_TYPE)':
#22 17.12 ../src/rcl_lifecycle_bindings.cpp:76:3: error: 'rcl_lifecycle_state_machine_options_t' was not declared in this scope; did you mean 'rcl_lifecycle_state_machine_init'?
#22 17.12    76 |   rcl_lifecycle_state_machine_options_t options =
#22 17.12       |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#22 17.12       |   rcl_lifecycle_state_machine_init
#22 17.12 In file included from ../src/rcl_lifecycle_bindings.cpp:29:
#22 17.12 ../src/rcl_lifecycle_bindings.cpp:83:33: error: 'options' was not declared in this scope; did you mean 'htons'?
#22 17.12    83 |                                &options),
#22 17.12       |                                 ^~~~~~~
#22 17.12 ../src/macros.hpp:22:16: note: in definition of macro 'CHECK_OP_AND_THROW_ERROR_IF_NOT_TRUE'
#22 17.12    22 |     if (lhs op rhs) {                                               \
#22 17.12       |                ^~~
#22 17.12 ../src/rcl_lifecycle_bindings.cpp:79:3: note: in expansion of macro 'THROW_ERROR_IF_NOT_EQUAL'
#22 17.12    79 |   THROW_ERROR_IF_NOT_EQUAL(RCL_RET_OK,
#22 17.12       |   ^~~~~~~~~~~~~~~~~~~~~~~~
#22 17.12 ../src/rcl_lifecycle_bindings.cpp: In lambda function:
#22 17.12 ../src/rcl_lifecycle_bindings.cpp:90:77: error: too few arguments to function 'rcl_ret_t rcl_lifecycle_state_machine_fini(rcl_lifecycle_state_machine_t*, rcl_node_t*, const rcl_allocator_t*)'
#22 17.12    90 |         rcl_ret_t ret = rcl_lifecycle_state_machine_fini(state_machine, node);
#22 17.12       |                                                                             ^
#22 17.12 In file included from ../src/rcl_lifecycle_bindings.cpp:18:
#22 17.12 /opt/ros/foxy/include/rcl_lifecycle/rcl_lifecycle.h:272:1: note: declared here
#22 17.12   272 | rcl_lifecycle_state_machine_fini(
#22 17.12       | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#22 17.22 make: Leaving directory '/root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/node_modules/rclnodejs/build'
#22 17.22 make: *** [rclnodejs.target.mk:374: Release/obj.target/rclnodejs/src/rcl_lifecycle_bindings.o] Error 1
#22 17.23 gyp ERR! build error
#22 17.23 gyp ERR! stack Error: `make` failed with exit code: 2
#22 17.23 gyp ERR! stack     at ChildProcess.onExit (/root/.nvm/versions/node/v12.20.1/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
#22 17.23 gyp ERR! stack     at ChildProcess.emit (events.js:314:20)
#22 17.23 gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
#22 17.23 gyp ERR! System Linux 4.15.0-142-generic
#22 17.23 gyp ERR! command "/root/.nvm/versions/node/v12.20.1/bin/node" "/root/.nvm/versions/node/v12.20.1/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
#22 17.23 gyp ERR! cwd /root/.nvm/versions/node/v12.20.1/lib/node_modules/ros2-web-bridge/node_modules/rclnodejs
#22 17.23 gyp ERR! node -v v12.20.1
#22 17.23 gyp ERR! node-gyp -v v5.1.0
#22 17.23 gyp ERR! not ok
#22 17.38 npm ERR! code ELIFECYCLE
#22 17.38 npm ERR! errno 1
#22 17.38 npm ERR! rclnodejs@0.19.0 install: `node-gyp rebuild`
#22 17.38 npm ERR! Exit status 1
#22 17.38 npm ERR!
#22 17.38 npm ERR! Failed at the rclnodejs@0.19.0 install script.
#22 17.38 npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
#22 17.39
#22 17.39 npm ERR! A complete log of this run can be found in:
#22 17.39 npm ERR!     /root/.npm/_logs/2021-05-26T04_01_09_629Z-debug.log
------
failed to solve with frontend dockerfile.v0: failed to build LLB: executor failed running [/bin/bash --login -c source /opt/ros/foxy/setup.bash     && npm install --global --unsafe-perm ros2-web-bridge]: runc did not terminate sucessfully
```

minggangw, 2021-05-26:
The latest v0.19.0 is not compatible with Foxy release, it runs on Galactic.

minggangw, 2021-05-26:
I'd like to have a new release (v0.19.1) that will support either Fox or Galactic, so users who use `ros2-web-bridge` don't have to pin the `rclnodejs` to v0.18.x when running on Foxy release, thanks!

wayneparrott, 2021-05-26:
Let me know if I can assist with the dual ros2 release support. I recall the biggest changes in rclnodejs were due to the lifecycle api changes in galactic.

minggangw, 2021-05-26:
That's it and I think we can do the same change as https://github.com/RobotWebTools/rclnodejs/commit/87375dedd4faaaa77376118d076864bb3d03111e does, @wayneparrott would you please offer a PR, thanks for helping!

commit ab69738d (2021-05-27): support foxy & galactic simultaneously (#793)

* support foxy & galactic simultaneously

ros_distro.js
- added ros galactic version id: 2105

rcl_lifecycle_binding.cpp
- uses ros galactic version id (2105) to select between foxy and
galactic lifecycle api

test-raw-pub-sub.js
- uses ros_distro.js to access ros version id
- for galactic version, test uses null-terminated string

test-node.js
- added small 100ms wait between publisher creation and test for
node.publisherCount() to avoid failing on slower systems.

fixes #792

* fixes lint issue
commit 3ec697a1 (2021-05-31): support foxy & galactic simultaneously (#793)

* support foxy & galactic simultaneously

ros_distro.js
- added ros galactic version id: 2105

rcl_lifecycle_binding.cpp
- uses ros galactic version id (2105) to select between foxy and
galactic lifecycle api

test-raw-pub-sub.js
- uses ros_distro.js to access ros version id
- for galactic version, test uses null-terminated string

test-node.js
- added small 100ms wait between publisher creation and test for
node.publisherCount() to avoid failing on slower systems.

fixes #792

* fixes lint issue
closed 2021-05-27 by minggangw