RobotWebTools/rclnodejs #794: sros2 support for rclnodejs

AnushaPulichintha, 2021-06-02:
Hi,
We are using sros2 for secure communication for our ros2 nodes. Recently we added ros node using rclnodejs library which communicates with our ros2 nodes. 

I tried to add the sros2 for the ros node running using rclnodejs library but it did not work. 

Does rclnodejs library support sros2?

Thanks in advance

minggangw, 2021-06-02:
We haven't tested with sros2 enabled before, I checked out the usage of sros2 and it seems that the encryption should be supported by the middle ware layer, would you please try (I haven't tried it myself, just guess we should pass the args to the `rcl`)
```js
rclnodejs.init(rclnodejs.Context.defaultContext(), ['--ros-args', '--enclave']);
```

AnushaPulichintha, 2021-06-02:
No, it did not work. Normally for ros2 we set some environment variables. I also set same variables for nodejs application.

```
export ROS_SECURITY_KEYSTORE=~/<ros2ws_dir>/keys
export ROS_SECURITY_ENABLE=true
export ROS_SECURITY_STRATEGY=Enforce
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export ROS_DOMAIN_ID=5
```
do you know I should pass any of those varaibles to init() method?

minggangw, 2021-06-03:
I'm not clear about the exact params needed for this case, I will have a try.