RobotWebTools/rclnodejs #817: Unable to receive data from different user on same machine

csmith-rtr, 2021-10-29:
**Description**
I'm trying to get a systemd service with `rclnodejs` to talk to another service using `rclcpp` in ros2 foxy. The services are run as different users. This causes an issue where the nodes cannot talk to each other and a resolution is presented [here](https://github.com/eProsima/Fast-DDS/issues/1750) using a custom xml configuration. Unfortunately, this doesn't appear to work for the `rclnodejs` process. The environment variable seems to be ignored or is functioning incorrectly somehow as the `rclnodejs` process is not receiving any data from `rclcpp` although I can verify that the data is being published via `ros2 topic echo`.

If I change the user of the services to match then `rclnodejs` is able to receive data.

Is there some way to verify that the environment variable is being passed correctly to the middleware? Or should `fastrtps` be taking care of that itself?

- Library Version: 0.18.2
- ROS Version: foxy
- Platform / OS: ubuntu

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

wayneparrott, 2021-10-29:
Thanks for this detailed report.  I ran few quick tests and determined:
1. With foxy and the default fastrtps rmw I can replicate the buggy behavior running separate pub-sub processes under different linux users, i.e., the subscriber never receives messages.
    
2. I can also confirm proper communications when using the [eclipse-cyclone dds rmw](https://docs.ros.org/en/foxy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html) on both foxy or galactic, i.e., subscriber receives all msgs.

>Is there some way to verify that the environment variable is being passed correctly to the middleware? Or should fastrtps be taking care of that itself?

My blink reaction is fastrtps should handle accessing the envar. Thinking out loud I'm wondering if your  xml file is correct and the envar is set correctly and fastrtps recognizes the config when run with something like the ros2 `talker` and `listener` examples.
```
ros2 run demo_nodes_cpp talker
```
And from a 2nd shell under a different user run:
```
ros2 run demo_nodes_cpp listener
```
If this all works then I would expect fastrtps to work in rclnodejs. Can you run a quick test using the example talker-listener example above and share the result?

Alternatively, switching to eclipse-cyclone is a quick install and envar setting for use with foxy. Let me know how you get on and if you stick with fastrtps and can't get the envar/xml combo to work we can help dig into what's goofed up.

wayneparrott, 2021-11-02:
I finally got around to testing foxy w/ fastrtps with a publisher and subscriber process running under 2 different linux user accounts respectively. I used both the rclnodejs publisher & subscriber example programs and the demo_nodes_cpp talker and listener examples for testing. Using the fastrtps configuration described here https://github.com/eProsima/Fast-DDS/issues/1750  I am able to receive messages. When I remove the xml config file from my environment no messages are received. 

With that, I believe this issue can be closed. I appreciate @csmith-rtr opening this issue and researching to identify the fastrtps configuration requirements. A follow-on action item for me will be to update the rclnodejs documentation to include the configuration requirements for this scenario.

csmith-rtr, 2021-11-02:
I think you're right. I've been able to get the publisher/subscriber examples working across users with different combinations of `rclnodejs` and `rclcpp`. Unfortunately I haven't been able to get it to work on my actual system, but at this point it seems like `rclnodejs` is doing the right thing, or isn't breaking the custom dds configuration somehow.

closed 2021-11-24 by csmith-rtr