RobotWebTools/rclnodejs #822: Cannot run rclnodejs with Node v12.22.8

eranroll, 2021-12-17:
I am trying to run rclnodejs as part of my nw application, but I get to following error:

```
node_modules/rclnodejs/build/Release/rclnodejs.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 79. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1086)
    at Module.load (internal/modules/cjs/loader.js:846)
    at Function.Module._load (internal/modules/cjs/loader.js:754)
    at Module.require (internal/modules/cjs/loader.js:888)
    at require (internal/modules/cjs/helpers.js:74)
    at bindings (home/parallels/configurator_mark3/node_modules/bindings/bindings.js:112)
    at Object.<anonymous> (home/parallels/configurator_mark3/node_modules/rclnodejs/lib/clock.js:17)
    at Module._compile (internal/modules/cjs/loader.js:995)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1041)
    at Module.load (internal/modules/cjs/loader.js:846)
```

I am using node v12.22.8 and npm install using npm v6.14.15

wayneparrott, 2021-12-17:
I'm not familiar with "nw application". My thought is that your app framework may be embed it's own version of node similar to electron. If so that would explain the NODE_MODULE_VERSION conflict.

node 12 = NODE_MODULE_VERSION  72
node 13 = NODE_MODULE_VERSION  79

At this time the rclnodejs v.0.20.1 can not run on version of node higher than v12. We have a node16 branch in development that we hope to publish very soon. 

Back to the problem please elaborate on the nw framework and we can work from there.

eranroll, 2021-12-18:
This is what I meant by an NW application: https://nwjs.io
I am using NW.js to run my app.

You suggestion made me look it up some more, and apparently the NW.js version I was using was based upon a newer version of node. I downgraded it to 0.41.3 which matches node12 and the issue was solved.

Thanks

closed 2021-12-18 by eranroll