RobotWebTools/rclnodejs #823: Typescript Numeric Arrays

csmith-rtr, 2021-12-17:
**Description**

Not sure if this is a bug or a question... Is there a reason numeric arrays are union types? Is there an option you can set somewhere that changes what the type actually is?

[image]

- Library Version:
- ROS Version:
- Platform / OS:

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

wayneparrott, 2021-12-17:
> Is there an option you can set somewhere that changes what the type actually is?

This is not a bug but by design. See the `Options` interface in node.d.ts. 
```typescript
/**
   * Configuration options when creating new Publishers, Subscribers,
   * Clients and Services.
   *
   * See {@link DEFAULT_OPTIONS}
   */
  interface Options<T = QoS | QoS.ProfileRef> {
    /**
     * A messages will use TypedArray if necessary, default: true.
     */
    enableTypedArray?: boolean;

    /**
     * Indicates messages are serialized, default: false.
     *
     * @remarks
     * See {@link Node#createSubscription | Node.createSubscription}
     */
    isRaw?: boolean;

    /**
     * ROS Middleware "quality of service" setting, default: QoS.profileDefault.
     */
    qos?: T;
  }

  /**
   * Default options when creating a Node, Publisher, Subscription, Client or Service
   *
   * ```ts
   * {
   *   enableTypedArray: true,
   *   qos: QoS.profileDefault,
   *   isRaw: false
   * }
   * ```
   */
  const DEFAULT_OPTIONS: Options;
```

DEFAULT_OPTIONS is the default parameter value when creating publishers, subscribers, services and actions. Here's the `createPublishers()` declaration that illustrates the optional `Options` parameter.

```
/**
     * Create a Publisher.
     *
     * @param typeClass - Type of message that will be published.
     * @param topic - Name of the topic the publisher will publish to.
     * @param options - Configuration options, see DEFAULT_OPTIONS
     * @returns New instance of Publisher.
     */
    createPublisher<T extends TypeClass<MessageTypeClassName>>(
      typeClass: T,
      topic: string,
      options?: Options
    ): Publisher<T>;
```

So you can customize the message serialization as shown:
```typescript
node.createPublisher(
  'std_msgs/msg/String', 
  'msg',
  {  enableTypedArray: true } // this is the default value and usually omitted in most of the examples
);
```

csmith-rtr, 2021-12-21:
Great! Thanks - sorry for the invalid bug.

closed 2021-12-21 by csmith-rtr