RobotWebTools/rclnodejs #842: Memory Leak when subscribing to messages containing UInt8Array

tytremblay, 2022-03-23:
**Description**
It seems memory is not getting released for subscribers subscribing to messages containing a UInt8Array.  If a sufficiently large array is published, the memory of the Node.js process can grow very quickly.

- Library Version: ^0.21.0
- ROS Version: foxy
- Platform / OS: Ubuntu 20.04.3 LTS (64-bit)

**Steps To Reproduce**
Attached is a small nodejs example of the issue I'm seeing.  In the example, I have set up a publisher to publish a large image every 100ms.  I've set up a subscriber to subscribe to that topic, and print `process.memoryUsage().rss` each time it receives a message.

The example can be run by running `npx ts-node server.ts` in a terminal window and `npx ts-node client.ts` in another window.

When both are running, you can see that the memory footprint of the node process running the client grows whenever a message is received.

**Expected Behavior**
Memory should not grow with each message

**Actual Behavior**
Memory grows with each message
[rosleak.zip](https://github.com/RobotWebTools/rclnodejs/files/8336169/rosleak.zip)

minggangw, 2022-03-24:
Thanks for your feedback, rclnodejs replaced one of its dependencies, `ref-napi`, which may cause this memory leak. Would you please help to test with [v0.20.1](https://www.npmjs.com/package/rclnodejs/v/0.20.1) & nodejs12 to see if you can still reproduce it? Thanks!

tytremblay, 2022-03-24:
I am unable to reproduce on v0.20.1 and node 12.  After 750 messages, the RSS size was ~200MB.  I repeated the same test with v0.21.0 and node 16 and, after 750 messages, the RSS size was ~950MB

minggangw, 2022-03-25:
Thanks for testing it! So I am pretty sure it is the change of `ref-napi` that leads to the memory leak, thanks for finding this critical issue! will fix it asap.

minggangw, 2022-04-02:
@tytremblay please try with the [latest 0.21.1](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.21.1) to check if the issue is gone, thanks!

minggangw, 2022-04-04:
@tytremblay good to know and thanks for your feedback!

xhy279, 2022-04-25:
Ty for the fix. Good job man.

commit 019d4431 (2022-04-02): Free memory allocated for typed arrays

This patch fixed the memory leak when using typed arrays on nodejs > 10

Fix: #842