RobotWebTools/rclnodejs #850: bool and bool_array should not be converted to strings in parameter msg

imcelroy, 2022-05-25:
**Description**

Parameter bool values are currently converted to strings (for example, `true` becomes `'true'`) when using the `Parameter.toParameterMessage()` method. These messages aren't parsed correctly when they are received by an rclcpp node and the `as_bool()` and `as_bool_array()` methods are used.

I have tested and it seems that the conversions here([parameter.js:L227-L234](https://github.com/RobotWebTools/rclnodejs/blob/222c5f0264199843b816c43f5ac8b73b0508e03c/lib/parameter.js#L227-L234)) aren't necessary, and the code works as expected if removed.

- Library Version: 0.21.1
- ROS Version: Galactic
- Platform / OS: Linux (Ubuntu 20.04)

**Steps To Reproduce**

- With rclnodejs:
```
//Create a parameter of type PARAMETER_BOOL or PARAMETER_BOOL_ARRAY
let parameters: Parameter[];
parameters.push(new Parameter('foo', ParameterType.PARAMETER_BOOL_ARRAY, [true, false, true]))

// Convert parameter to msg
const setParametersRequest: rcl_interfaces.srv.SetParameters_Request = {
      parameters: parameters.map(function (e) {
        return e.toParameterMessage();
      }),
    };

// Send request using the client Client<'rcl_interfaces/srv/SetParametersAtomically'> or Client<'rcl_interfaces/srv/SetParameters'>
client.sendRequest(setParametersRequest, (resp) => {
          console.log('resp: ', resp)
        });
```

- With rclcpp:
```
rcl_interfaces::msg::SetParametersResult parameters_callback(
  const vector<rclcpp::Parameter> & parameters)
{
  rcl_interfaces::msg::SetParametersResult result;
  result.successful = true;
  for (const rclcpp::Parameter & param : parameters) {
      if (param.get_name() == "dofs_activations") {
        vector<bool> foo = param.as_bool_array();
        RCLCPP_INFO_STREAM(
          get_node()->get_logger(), "Received foo : " <<
            foo[0] << " " <<
            foo[1] << " " <<
            foo[2];
   }
return result;
 }
```

**Expected Behavior**
`1 0 1` should be outputted in the terminal.

**Actual Behavior**
`1 1 1` is actually outputted in the terminal.

closed 2022-05-27 by minggangw

imcelroy, 2022-05-30:
@minggangw @wayneparrott Thanks for accepting the PR. Any chance we could get this fix in a new npm release asap?

minggangw, 2022-05-30:
I think we will have a new release targeting against the latest Humble Hawksbill soon.

imcelroy, 2022-05-30:
@minggangw Ok great, will the release targeting Humble still be compatible with Galactic?

minggangw, 2022-05-31:
@imcelroy Unlike the [rclcpp](https://github.com/ros2/rclcpp)/[rclpy](https://github.com/ros2/rclpy) clients that will be released with the ROS2 packages, so you don't need to worry about the compatibility. rclnodejs is a stand-alone nodejs package, and as you know ROS2 has many releases in parallel and their patch releases, which makes it difficult to maintain the compatibility. We did try to leverage e.g., build flag, to cover different ROS2 releases, but usually, we only verify the binary with our tests on the latest stable ROS2 due to limited resources.

imcelroy, 2022-05-31:
Thank you for the explanation, that makes sense.

commit 688979b5 (2022-05-27): do not convert bool to string of bool for parameter msg (#851)

Public API Changes

None

Description

Remove conversion of bool values to strings for ParameterType.PARAMETER_BOOL and ParameterType.PARAMETER_BOOL_ARRAY in Parameter.toParameterMessage(). The rclcpp methods Parameter::as_bool() and Parameter::as_bool_array() do not convert these messages correctly if they contain bools as strings.

fixes: #850