RobotWebTools/rclnodejs #854: Replace appveyor with Github action/workflow for Windows builds

wayneparrott, 2022-05-26:
The project recently switch to using Github Actions for Linux builds. This PR proposes to replace the Appveyor CI with Github Actions similar to the Linux build model.

See related issues: #841, #834

wayneparrott, 2022-05-26:
Currently blocked by bug https://github.com/ros-tooling/setup-ros/issues/491

wayneparrott, 2022-06-02:
[setup-ros v0.3.4](https://github.com/marketplace/actions/setup-ros-environment) has been released. We should be able to resume work on this. I'm hoping that #856 will reduce many of the past challenges with windows ci.

minggangw, 2022-06-03:
Good news! Thanks for keeping looking at this issue.

minggangw, 2022-06-05:
I have stopped the [appveyor](https://ci.appveyor.com/project/minggangw/rclnodejs) for rclnodejs, as it's always failing :(

minggangw, 2022-06-13:
I think we can mark this as resolved, thanks!

closed 2022-06-13 by minggangw