RobotWebTools/rclnodejs #856: Remove cases that depends on test/rclnodejs_test_msgs under \test

minggangw, 2022-05-31:
As these tests become unstable and cause the CI failures frequently, we decide to remove the C++ project used to verify some interactions with other ROS2 clients.

minggangw, 2022-06-01:
For the test-non-primitive-static-arrary.js test, I don't find any `.msg` that uses a `Time` array in the current ROS2  release, so we may have to find something equivalent, maybe the [JoyFeedbackArray.msg](
https://github.com/ros2/common_interfaces/blob/618473bef14632e2b3769049df00f69563e4c018/sensor_msgs/msg/JoyFeedbackArray.msg) is suitable to replace? They are both kinds of primitives array, but we have to re-implement the case if we want to replace with it.

wayneparrott, 2022-06-02:
@minggangw let me confirm that addressing this issue involves the following actions: 
1. remove test/rclnodejs_test_msgs/ folder
2. remove script/compile_tests.js
- note this script currently compiles: publisher_msg.cpp, subscription_msg.cpp, listener.cpp
4. remove the `pretest` script in package.json that currenctly calls compile_test.js script
5. use fibonacci msg from test_interface_files pkg in place of rclnodejs_test_msgs pkg in the following tests:
test-action-graph.js
test-action-client.js
test-action-server.js
test/types/main.ts
6. clean up hacks in linux workflows to get around frequent env challenges when the rclnodejs_test_msgs pkg was built during testing
7. TBD - replace test-non-primitive-static-array.js with an equivalent test using a fixed size Array<complex type>

Q's: 
1. do we continue to support compiling the test/cpp/ project?
if no, 
remove:
 test/cpp
 test/py
 test-cross-lang.js
 test-multi-nodes.js

I'll be glad to make any/all changes if you want to assign this issue to me.

minggangw, 2022-06-03:
> remove test/rclnodejs_test_msgs/ folder

Yes
> remove script/compile_tests.js
> note this script currently compiles: publisher_msg.cpp, subscription_msg.cpp, listener.cpp

[add_two_ints_client.cpp](https://github.com/RobotWebTools/rclnodejs/blob/325848bd85237f2d754661755e0bb9aeb6f3c060/test/cpp/add_two_ints_client.cpp) will also be removed, 

> remove the pretest script in package.json that currenctly calls compile_test.js script

Yes
> use fibonacci msg from test_interface_files pkg in place of rclnodejs_test_msgs pkg in the following tests:
> test-action-graph.js
> test-action-client.js
> test-action-server.js
> test/types/main.ts

We can depend on the CI to check if we miss some test cases, as there are so many.
 
> clean up hacks in linux workflows to get around frequent env challenges when the rclnodejs_test_msgs pkg was built during testing

Yes, finally we can get rid of it :)
> TBD - replace test-non-primitive-static-array.js with an equivalent test using a fixed size Array

If it's not easy to find an alternative, we can add this case into the [blocklist](https://github.com/RobotWebTools/rclnodejs/blob/develop/test/blocklist.json)

>do we continue to support compiling the test/cpp/ project?

I suggest we will not to add them back in the future, we will focus on the JS case only.

minggangw, 2022-06-03:
@wayneparrott I have re-assigned this issue to you, please take your time, it's not urgent, thanks and stay healthy!

PR #860 (2022-06-13): rm rclnodejs_test_msgs, compile script, cpp & py dirs, consolidates linux workflows [#856]
The main objective of this PR is to remove the troublesome compile_scripts.js, rclnodejs_test_msgs pkg and related tests and cpp/py example clients from the repo.

1. removed test/rclnodejs_test_msgs and related tests
- removed scripts/compile_tests.js and package.json script references
- replaced reference to Fibonacci action msgs in test-action-*.js and ts
types test files with equivalent msgs in test_msgs package

2 - discontinued testing compatibility with cpp and py rcl ros clients
- removed test/cp [... cut]
commit e5ddff03 (2022-06-13): rm rclnodejs_test_msgs, compile script, cpp & py dirs, consolidates linux workflows [#856] (#860)

* rm rclnodejs_test_msgs, compile script, cpp & py dir, improved linux wf
1. removed test/rclnodejs_test_msgs and related tests
- removed scripts/compile_tests.js and package.json script references
- replaced reference to Fibonacci action msgs in test-action-*.js and ts
types test files with equivalent msgs in test_msgs package

2 - discontinued testing compatibility with cpp and py rcl ros clients
- removed test/cpp/ folder
- removed test/py/ folder

3 - Replace test-non-primitive-static-array.j [... cut]
closed 2022-06-13 by minggangw