RobotWebTools/rclnodejs #857: Preparing rclnodejs v0.21.2 release

minggangw, 2022-06-01:
We are going to release [v0.21.2](https://github.com/RobotWebTools/rclnodejs/projects/19) with latest bug fix targeting on [Humble Hawksbill](https://github.com/ros2/ros2/releases/tag/release-humble-20220523)

minggangw, 2022-06-01:
@wayneparrott please add things we want to land on the next rclnodejs release for https://github.com/RobotWebTools/rclnodejs/projects/19 if I'm missing something important, thanks!

ejalaa12, 2022-06-02:
@imcelroy and I are trying to work on a possible solution for the following bug https://github.com/RobotWebTools/rclnodejs/issues/858.
Can you please take a look at it and see if this is important enough to include in this release ?

minggangw, 2022-06-02:
@ejalaa12 of course, we can wait for the fix to be landed and then roll out the release, I will take a look at that issue, thanks!

minggangw, 2022-06-13:
I think we are good to go here, will roll out this version soon, thanks to all contributors for making this happen!

centercao, 2022-06-16:
please support node.js 18.xxx

minggangw, 2022-06-16:
@centercao do you meet any problems when using with nodejs 18?

minggangw, 2022-06-16:
[0.21.2](https://www.npmjs.com/package/rclnodejs/v/0.21.2) rolls out! Thanks for everybody's contributions!

closed 2022-06-16 by minggangw