RobotWebTools/rclnodejs #865: Investigate the use of prebuilt binaries to simplify installation requirements & experience

wayneparrott, 2022-07-18:
Can we simplify the rclnodejs installation requirements? Specifically the compilation for C-lang native api.

To answer this question I have created this tracking issue to investigate the use of [prebuild](https://github.com/prebuild/prebuild) for packaging and distributing compiled binaries.

pabloinigoblasco, 2023-11-14:
Is there any example url about how to do this in one "rclnodejs regular package"?

minggangw, 2023-11-15:
@pabloinigoblasco do you mean you want `rclnodejs` to be a prebuild component in your project?

pabloinigoblasco, 2023-11-15:
Hello. 
Well I guess the idea would be having "ready to use debian packages".
When you install a ros package via apt it is IMHO ready to use because all the dependencies are installed and nothing else is required to be done. For example:
```
apt install ros-humble-myowesome-nodejs-package
source /opt/ros/humble/setup.bash
ros2 launch myowesome_nodejs_package example.launch.py
```
The only a ros package developer have to do is usually add the system or ros dependency in the package.xml file.

I have some doubts that would work for nodejs packages (created by rclnodejs-cli).  If this is possible, great, we are fine.

However, I am thinking that the nodejs apt package in ubuntu jammy is 12.22.9 but rclnodejs requires at least 16.x.x , am I right?

If so, I understand that you could install nvm or nodejs manually, etc, but that is not the point.

Do you know if this problem is solved?

If it isnt, I think there could be some "tradeoff solution"  based on some debian package post-install-scripts.