RobotWebTools/rclnodejs #876: Error during install: ‘::memchr’ has not been declared

chfritz, 2022-09-01:
**Description**

When I try to install using `npm install rclnodejs`, it fails with:
```
error: ‘::memchr’ has not been declared
```
(full log below).

- Library Version: latest from npm (0.21.3)
- ROS Version: ros2 rolling
- Platform / OS: Ubuntu 20.04.5

```
> npm version
{
  x: '1.0.0',
  npm: '8.15.0',
  node: '16.17.0',
  v8: '9.4.146.26-node.22',
  uv: '1.43.0',
  zlib: '1.2.11',
  brotli: '1.0.9',
  ares: '1.18.1',
  modules: '93',
  nghttp2: '1.47.0',
  napi: '8',
  llhttp: '6.0.7',
  openssl: '1.1.1q+quic',
  cldr: '41.0',
  icu: '71.1',
  tz: '2022a',
  unicode: '14.0',
  ngtcp2: '0.1.0-DEV',
  nghttp3: '0.1.0-DEV'
}
```

**Steps To Reproduce**

`npm install rclnodejs`

**Expected Behavior**

For it to install as described in the readme.

**Actual Behavior**

```
> npm install rclnodejs
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm ERR! code 1
npm ERR! path /tmp/x/node_modules/rclnodejs
npm ERR! command failed
npm ERR! command sh /tmp/install-9ff58467.sh
npm ERR! make: Entering directory '/tmp/x/node_modules/rclnodejs/build'
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/addon.o
npm ERR! make: Leaving directory '/tmp/x/node_modules/rclnodejs/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.0.0
npm ERR! gyp info using node@16.17.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.8.10 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
[... 106 lines cut ...]
npm ERR!       |          ^~~~~~
npm ERR!       |          memchr
npm ERR! ../src/addon.cpp: In function ‘void InitModule(v8::Local<v8::Object>)’:
npm ERR! ../src/addon.cpp:48:16: error: ‘strstr’ was not declared in this scope; did you mean ‘std::strstr’?
npm ERR!    48 |     auto end = strstr(prog_name, " -");
npm ERR!       |                ^~~~~~
npm ERR!       |                std::strstr
npm ERR! In file included from ../../nan/nan.h:64,
npm ERR!                  from ../src/addon.cpp:15:
npm ERR! /usr/include/c++/9/cstring:118:3: note: ‘std::strstr’ declared here
npm ERR!   118 |   strstr(char* __s1, const char* __s2)
npm ERR!       |   ^~~~~~
npm ERR! make: *** [rclnodejs.target.mk:162: Release/obj.target/rclnodejs/src/addon.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Linux 5.15.0-46-generic
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /tmp/x/node_modules/rclnodejs
npm ERR! gyp ERR! node -v v16.17.0
npm ERR! gyp ERR! node-gyp -v v9.0.0
npm ERR! gyp ERR! not ok
```

chfritz, 2022-09-01:
I've discovered that `npm install rclnodejs@0.21.1` works, so it seems that the issue was introduced recently.

minggangw, 2022-09-02:
Thanks for reporting the issue, through the error log, it seems that it happens when compiling the nan, do you try with Humble Hawksbill?

chfritz, 2022-09-02:
I haven't -- I'm on ubuntu 20 -- but the `nan.h` file is not from the OS but part of the npm nan package: https://github.com/nodejs/nan/blob/main/nan.h. So I don't think changing the ROS2 release would make a difference.

minggangw, 2022-09-03:
Yes, the header comes from nan package, theoretically, the behavior should be the same on different rclnodejs versions. I also checked the code change between these 2 versions, and it seems nothing special (we don't upgrade the nan version number). I will try to build it on my dev machine, but I believe it's caused by some toolchain env problem.

meropis, 2022-09-05:
I'm also having a similar issue
Ubuntu 22.04 LTS
Humble Hawkesbill

minggangw, 2022-09-06:
Would you try to upgrade your GCC to 11 and see if the failure is still there? I verified with the following configuration and it works fine:

- nodejs:  v16.16.0
- g++: 11.2.0
- OS: Ubuntu 22.04LTS

Meanwhile, looking through the latest build on [actions](https://github.com/RobotWebTools/rclnodejs/actions), no failures were found.

wayneparrott, 2022-09-06:
I noticed your env includes:
> Platform / OS: Ubuntu 20.04.5

My understanding as of the Humble release, includes latest Rolling release, that Ubuntu 22 (jammy) is required. As a quick test I set up a new dev env with Ubuntu 22.04.01 (lts), Node 16 (lts) and rclnodejs from npm. No issues observed with rclnodejs installation and running simple node was successful. 

Hope this helps.

chfritz, 2022-09-07:
g++-11 is not available on 20.04 so I can't try that. I've also noticed that there is no build status for galactic right now (see Readme), so I assume that combination is not currently being tested? Another question: is there a reason why the repo doesn't include a `package-lock.json`? Without that, it's not deterministic which version of the dependencies was used in current and past npm releases. We could find out, of course, by inspecting the npm bundle, but wouldn't be able to reproduce it from the 
 tag commit alone.

wayneparrott, 2022-09-07:
> I've also noticed that there is no build status for galactic right now (see Readme), so I assume that combination is not currently being tested?

Opps! Thx for calling this out. Will get the status indicator fixed asap. The os/ros compatibility matrix are completing successfully for linux. https://github.com/RobotWebTools/rclnodejs/actions/runs/3009191104

Looks like a dependent action for setting up ros distros on windows is failing (again). Will give it a look asap. 
https://github.com/RobotWebTools/rclnodejs/runs/8232819051?check_suite_focus=true

minggangw, 2022-10-06:
@chfritz @meropis finally, I hit this issue on 20.04 with g++9.4, and I have a workaround to mitigate this weird issue (it cannot be reproduced with 22.04...), please remove the following section from your [`binding.gyp`](https://github.com/RobotWebTools/rclnodejs/blob/develop/binding.gyp)
```
            'conditions': [
              [
                'ros_version > 2105', # Humble, Rolling, ...
                {
                  'include_dirs': 
                  [
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/ ') + '/include/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcl/ ') + '/include/rcl')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcutils/ ') + '/include/rcutils/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rmw/ ') + '/include/rmw/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcl_yaml_param_parser/ ') + '/include/rcl_yaml_param_parser/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rosidl_typesupport_interface/ ') + '/include/rosidl_typesupport_interface/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcl_action/ ') + '/include/rcl_action/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/action_msgs/ ') + '/include/action_msgs/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/unique_identifier_msgs/ ') + '/include/unique_identifier_msgs/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/builtin_interfaces/ ') + '/include/builtin_interfaces/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcl_lifecycle/ ') + '/include/rcl_lifecycle/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/lifecycle_msgs/ ') + '/include/lifecycle_msgs/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rosidl_runtime_c/ ') + '/include/rosidl_runtime_c/')\")",
                  ],
                }
              ],
```

I will submit a PR to fix it and prepare a hotfix release, thanks!

chfritz, 2023-07-10:
Sorry for the delay in testing this.

Removing the lines in `binding.gyp` indeed get me past the `::memchr` error. But I'm getting:

```
> npm i

> rclnodejs@0.22.2 install
> npm run rebuild

> rclnodejs@0.22.2 rebuild
> npm run clean && node-gyp -j 16 rebuild

> rclnodejs@0.22.2 clean
> node-gyp clean && rimraf ./generated

make: Entering directory '/tmp/k/rclnodejs/build'
  CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
  CXX(target) Release/obj.target/rclnodejs/src/handle_manager.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_action_bindings.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_bindings.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_handle.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_lifecycle_bindings.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_utilities.o
  CXX(target) Release/obj.target/rclnodejs/src/shadow_node.o
../src/rcl_bindings.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE rclnodejs::CreateSubscription(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rcl_bindings.cpp:715:23: error: ‘rcl_subscription_options_set_content_filter_options’ was not declared in this scope
  715 |       rcl_ret_t ret = rcl_subscription_options_set_content_filter_options(
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

chfritz, 2023-07-10:
More details: when I use v0.21.2 then removing the lines from binding.gyp works on Ubuntu 20 but fails on Ubuntu 22.

chfritz, 2023-07-10:
Ok, so I've tested a whole matrix of combinations and finally realized that rolling on Ubuntu 20 is a different version from rolling on Ubuntu 22 (and the condition in bindings.gyp isn't able to distinguish them) . I'll try to come up with a better condition.

closed 2025-06-30 by minggangw