RobotWebTools/rclnodejs #880: Install error on Ubuntu 20.04 with Ros2 Humble

MMM-Jasmin, 2022-09-22:
HI,
i am trying to install install this package on a nvidia AGX Orin and I am facing compiling errors.
To get more info I used "npm install --loglevel verbose" on this setup:

- Library Version:  0.21.3
- ROS Version: humble
- Platform / OS: Ubuntu 20.04 but the linux for tegra version

and this is the output:

```
> rclnodejs@0.21.3 install
> node-gyp rebuild

gyp info it worked if it ends with ok
gyp verb cli [
gyp verb cli   '/usr/bin/node',
gyp verb cli   '/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js',
gyp verb cli   'rebuild'
gyp verb cli ]
gyp info using node-gyp@9.0.0
gyp info using node@16.17.0 | linux | arm64
gyp verb command rebuild []
gyp verb command clean []
gyp verb clean removing "build" directory
gyp verb command configure []
gyp verb find Python Python is not set from command line or npm configuration
gyp verb find Python Python is not set from environment variable PYTHON
gyp verb find Python checking if "python3" can be used
gyp verb find Python - executing "python3" to get executable path
gyp verb find Python - executable path is "/usr/bin/python3"
gyp verb find Python - executing "/usr/bin/python3" to get version
gyp verb find Python - version is "3.8.10"
gyp info find Python using Python version 3.8.10 found at "/usr/bin/python3"
gyp verb get node dir no --target version specified, falling back to host node version: 16.17.0
gyp verb command install [ '16.17.0' ]
gyp verb install input version string "16.17.0"
gyp verb install installing version: 16.17.0
gyp verb install --ensure was passed, so won't reinstall if already installed
gyp verb install version is already installed, need to check "installVersion"
gyp verb got "installVersion" 9
gyp verb needs "installVersion" 9
gyp verb install version is good
gyp verb get node dir target node version installed: 16.17.0
gyp verb build dir attempting to create "build" dir: /mnt/storage/test-area/rclnodejs/build
gyp verb build dir "build" dir needed to be created? Yes
gyp verb build/config.gypi creating config file
gyp verb build/config.gypi writing out config file: /mnt/storage/test-area/rclnodejs/build/config.gypi
gyp verb config.gypi checking for gypi file: /mnt/storage/test-area/rclnodejs/config.gypi
gyp verb common.gypi checking for gypi file: /mnt/storage/test-area/rclnodejs/common.gypi
gyp verb gyp gyp format was not specified; forcing "make"
gyp info spawn /usr/bin/python3
gyp info spawn args [
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/mnt/storage/test-area/rclnodejs/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/jasmin/.cache/node-gyp/16.17.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/jasmin/.cache/node-gyp/16.17.0',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/home/jasmin/.cache/node-gyp/16.17.0/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/mnt/storage/test-area/rclnodejs',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp verb command build []
gyp verb build type Release
gyp verb architecture arm64
gyp verb node dev dir /home/jasmin/.cache/node-gyp/16.17.0
gyp verb `which` succeeded for `make` /usr/bin/make
gyp info spawn make
gyp info spawn args [ 'V=1', 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/mnt/storage/test-area/rclnodejs/build'
make: execvp: printf: Argument list too long
make: *** [rclnodejs.target.mk:8922: Release/obj.target/rclnodejs/src/addon.o] Error 127
make: Leaving directory '/mnt/storage/test-area/rclnodejs/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
gyp ERR! System Linux 5.10.104-tegra
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /mnt/storage/test-area/rclnodejs
gyp ERR! node -v v16.17.0
gyp ERR! node-gyp -v v9.0.0
gyp ERR! not ok
npm timing command:install Completed in 5540ms
npm verb stack Error: command failed
npm verb stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/lib/index.js:63:27)
npm verb stack     at ChildProcess.emit (node:events:513:28)
npm verb stack     at maybeClose (node:internal/child_process:1093:16)
npm verb stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)
npm verb pkgid rclnodejs@0.21.3
npm verb cwd /mnt/storage/test-area/rclnodejs
npm verb Linux 5.10.104-tegra
npm verb node v16.17.0
npm verb npm  v8.15.0
npm ERR! code 1
npm ERR! path /mnt/storage/test-area/rclnodejs
npm ERR! command failed
npm ERR! command sh /tmp/install-6c0a1c6f.sh
npm verb exit 1
npm timing npm Completed in 5654ms
npm verb code 1

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/jasmin/.npm/_logs/2022-09-22T10_21_48_410Z-debug-0.log

```

Sadly i am not an expert on nodejs and hopefully someone can help me
Kind regards,
Nils

minggangw, 2022-09-23:
@MMM-Jasmin thanks for your feedback, I have very limited experience on the arm64 platform. What I can think about is that this issue is caused by the toolchain, do you use GCC and what's the version of it?

MMM-Jasmin, 2022-09-23:
Hi minggangw, thanks for the help!
So i tried gcc/g++ 8,9 and 10 and before I went a bit further down into the build process.
The created Makefile is trying to compile against everything from ros2 and the command is extremely long.
If you try to call it you get the g++/execv error:

g++: fatal error: cannot execute ‘/usr/lib/gcc/aarch64-linux-gnu/10/cc1plus’: execv: Argument list too long

Tests to increase limits for example with ulimit -s 652768 are also not working.
Maybe the Makefile should be more specific what to compile with.
Any idea?

minggangw, 2022-09-30:
Sorry for the delay, I have the following platforms
- Ubuntu 20.04 + g++ 9.4
- Ubuntu 22.01 + g++ 11.2

Both work fine, as you mentioned, maybe you need special flags that are different from the x86 desktop. I don't have any idea so far.

MMM-Jasmin, 2022-10-04:
Hey @minggangw,
i digged a little bit deeper, but still do not fully understand node-gyp.
In the "binding.gyp" is a array of "include_dirs" made namely:

>  'ros_version > 2105', # Humble, Rolling, ...
                {
                  'include_dirs':
                  [
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/ ') + '/include/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcl/ ') + '/include/rcl')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcutils/ ') + '/include/rcutils/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rmw/ ') + '/include/rmw/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcl_yaml_param_parser/ ') + '/include/rcl_yaml_param_parser/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rosidl_typesupport_interface/ ') + '/include/rosidl_typesupport_interface/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcl_action/ ') + '/include/rcl_action/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/action_msgs/ ') + '/include/action_msgs/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/unique_identifier_msgs/ ') + '/include/unique_identifier_msgs/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/builtin_interfaces/ ') + '/include/builtin_interfaces/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rcl_lifecycle/ ') + '/include/rcl_lifecycle/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/lifecycle_msgs/ ') + '/include/lifecycle_msgs/')\")",
                    "<!@(node -e \"console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/rosidl_runtime_c/ ') + '/include/rosidl_runtime_c/')\")",
                  ],
                }

It is based on "AMENT_PREFIX_PATH", which is already long. The result combination is saved in "build/rclnodejs.target.mk":

> INCS_Debug := \
        -I/home/jasmin/.cache/node-gyp/16.17.0/include/node \
        -I/home/jasmin/.cache/node-gyp/16.17.0/src \
        -I/home/jasmin/.cache/node-gyp/16.17.0/deps/openssl/config \
        -I/home/jasmin/.cache/node-gyp/16.17.0/deps/openssl/openssl/include \
        -I/home/jasmin/.cache/node-gyp/16.17.0/deps/uv/include \
        -I/home/jasmin/.cache/node-gyp/16.17.0/deps/zlib \
        -I/home/jasmin/.cache/node-gyp/16.17.0/deps/v8/include \
        -I$(srcdir)/. \
        -I$(srcdir)/node_modules/nan \
        -I/ros/rqt_bag_plugins/include \
        -I/ros/rosbag2/include \
        -I/ros/rosbag2_compression_zstd/include \
        -I/ros/zstd_vendor/include \
        -I/ros/rviz_visual_testing_framework/include \
        -I/ros/rviz2/include \
        -I/ros/rviz_default_plugins/include \
        -I/ros/rviz_common/include \
        -I/ros/rqt_bag/include \
        -I/ros/ros2bag/include \
        -I/ros/rosbag2_py/include \
        -I/ros/rosbag2_transport/include \
        -I/ros/rosbag2_performance_benchmarking/include \
........................................

I counted the includes and it tries to link against 4220 entries.
And most of them are not valid to begin with.
Funny that its is working for you, the g++ call is 196873 bytes long :/
Any ideas how the amount of includes could be reduced?

Kind regards,
Nils

minggangw, 2022-10-06:
Another way that I think should be workable is:

Change the `include_dirs` part manually to point to the actual `ros` path on your machine, like:
```
            'include_dirs': 
            [
              "opt/ros/rolling/include/"
            ]
```

wayneparrott, 2022-10-06:
@minggangw @MMM-Jasmin 
The long include path is excessive, yikes! It arose to handle differences in the file header structure between galactic/foxy distros and humble. I share this so you'll be aware of this diff if you are working across ros2 distros. @minggangw  perhaps we can address this in a more efficient manner in the near future. Also, the Humble distro is based on Ubuntu 22. I've seen some refs to articles for building on Ubuntu 20 but not done it myself to identify the issues.

MMM-Jasmin, 2022-10-07:
Hi @minggangw  @wayneparrott
i will try to set the includes manually today and maybe i can compile it then.

@minggangw i guess you installed the rebuild ros2 package?
My AMENT_PREFIX_PATH equals to 

`/opt/dev/ros2_humble/install/rqt_bag_plugins:/opt/dev/ros2_humble/install/rosbag2:/opt/dev/ros2_humble/install/rosbag2_compression_zstd:/opt/dev/ros2_humble/install/zstd_vendor:/opt/dev/ros2_humble/install/rviz_visual_testing_framework:/opt/dev/ros2_humble/install/rviz2:/opt/dev/ros2_humble/install/rviz_default_plugins:/opt/dev/ros2_humble/install/rviz_common:/opt/dev/ros2_humble/install/rqt_bag:/opt/dev/ros2_humble/install/ros2bag:/opt/dev/ros2_humble/install/rosbag2_py:/opt/dev/ros2_humble/install/rosbag2_transport:/opt/dev/ros2_humble/install/rosbag2_performance_benchmarking:/opt/dev/ros2_humble/install/rosbag2_compression:/opt/dev/ros2_humble/install/rosbag2_cpp:/opt/dev/ros2_humble/install/rosbag2_storage_default_plugins:/opt/dev/ros2_humble/install/rosbag2_storage:/opt/dev/ros2_humble/install/image_common:/opt/dev/ros2_humble/install/camera_info_manager:/opt/dev/ros2_humble/install/camera_calibration_parsers:/opt/dev/ros2_humble/install/yaml_cpp_vendor:/opt/dev/ros2_humble/install/interactive_markers:/opt/dev/ros2_humble/install/common_interfaces:/opt/dev/ros2_humble/install/visualization_msgs:/opt/dev/ros2_humble/install/dummy_robot_bringup:/opt/dev/ros2_humble/install/robot_state_publisher:/opt/dev/ros2_humble/install/kdl_parser:/opt/dev/ros2_humble/install/urdf:/opt/dev/ros2_humble/install/urdf_parser_plugin:/opt/dev/ros2_humble/install/turtlesim:/opt/dev/ros2_humble/install/tracetools_test:/opt/dev/ros2_humble/install/tracetools_launch:/opt/dev/ros2_humble/install/topic_monitor:/opt/dev/ros2_humble/install/geometry2:/opt/dev/ros2_humble/install/tf2_tools:/opt/dev/ros2_humble/install/tf2_sensor_msgs:/opt/dev/ros2_humble/install/test_tf2:/opt/dev/ros2_humble/install/tf2_kdl:/opt/dev/ros2_humble/install/tf2_geometry_msgs:/opt/dev/ros2_humble/install/examples_tf2_py:/opt/dev/ros2_humble/install/tf2_ros_py:/opt/dev/ros2_humble/install/tf2_eigen:/opt/dev/ros2_humble/install/tf2_bullet:/opt/dev/ros2_humble/install/tf2_ros:/opt/dev/ros2_humble/install/tf2_py:/opt/dev/ros2_humble/install/tf2_msgs:/opt/dev/ros2_humble/install/test_msgs:/opt/dev/ros2_humble/install/sros2_cmake:/opt/dev/ros2_humble/install/ros2cli_common_extensions:/opt/dev/ros2_humble/install/sros2:/opt/dev/ros2_humble/install/rqt_topic:/opt/dev/ros2_humble/install/rqt_srv:/opt/dev/ros2_humble/install/rqt_shell:/opt/dev/ros2_humble/install/rqt_service_caller:/opt/dev/ros2_humble/install/rqt_reconfigure:/opt/dev/ros2_humble/install/rqt_py_console:/opt/dev/ros2_humble/install/rqt_publisher:/opt/dev/ros2_humble/install/rqt_plot:/opt/dev/ros2_humble/install/rqt_action:/opt/dev/ros2_humble/install/rqt_msg:/opt/dev/ros2_humble/install/rqt_console:/opt/dev/ros2_humble/install/rqt:/opt/dev/ros2_humble/install/rqt_py_common:/opt/dev/ros2_humble/install/rqt_graph:/opt/dev/ros2_humble/install/rqt_gui_py:/opt/dev/ros2_humble/install/rqt_gui:/opt/dev/ros2_humble/install/ros_testing:/opt/dev/ros2_humble/install/ros2trace:/opt/dev/ros2_humble/install/ros2topic:/opt/dev/ros2_humble/install/ros2test:/opt/dev/ros2_humble/install/ros2component:/opt/dev/ros2_humble/install/ros2param:/opt/dev/ros2_humble/install/ros2lifecycle:/opt/dev/ros2_humble/install/ros2service:/opt/dev/ros2_humble/install/ros2run:/opt/dev/ros2_humble/install/ros2launch:/opt/dev/ros2_humble/install/ros2pkg:/opt/dev/ros2_humble/install/ros2node:/opt/dev/ros2_humble/install/ros2multicast:/opt/dev/ros2_humble/install/ros2interface:/opt/dev/ros2_humble/install/ros2doctor:/opt/dev/ros2_humble/install/ros2cli_test_interfaces:/opt/dev/ros2_humble/install/ros2action:/opt/dev/ros2_humble/install/ros2cli:/opt/dev/ros2_humble/install/quality_of_service_demo_py:/opt/dev/ros2_humble/install/quality_of_service_demo_cpp:/opt/dev/ros2_humble/install/image_transport:/opt/dev/ros2_humble/install/message_filters:/opt/dev/ros2_humble/install/lifecycle_py:/opt/dev/ros2_humble/install/launch_testing_ros:/opt/dev/ros2_humble/install/demo_nodes_cpp:/opt/dev/ros2_humble/install/composition:/opt/dev/ros2_humble/install/launch_ros:/opt/dev/ros2_humble/install/laser_geometry:/opt/dev/ros2_humble/install/examples_rclpy_pointcloud_publisher:/opt/dev/ros2_humble/install/examples_rclpy_minimal_subscriber:/opt/dev/ros2_humble/install/examples_rclpy_minimal_service:/opt/dev/ros2_humble/install/examples_rclpy_minimal_publisher:/opt/dev/ros2_humble/install/examples_rclpy_minimal_client:/opt/dev/ros2_humble/install/examples_rclpy_minimal_action_server:/opt/dev/ros2_humble/install/examples_rclpy_minimal_action_client:/opt/dev/ros2_humble/install/examples_rclpy_guard_conditions:/opt/dev/ros2_humble/install/examples_rclpy_executors:/opt/dev/ros2_humble/install/demo_nodes_py:/opt/dev/ros2_humble/install/action_tutorials_py:/opt/dev/ros2_humble/install/rclpy:/opt/dev/ros2_humble/install/examples_rclcpp_minimal_action_server:/opt/dev/ros2_humble/install/examples_rclcpp_minimal_action_client:/opt/dev/ros2_humble/install/action_tutorials_cpp:/opt/dev/ros2_humble/install/rclcpp_action:/opt/dev/ros2_humble/install/rcl_action:/opt/dev/ros2_humble/install/examples_rclcpp_wait_set:/opt/dev/ros2_humble/install/examples_rclcpp_minimal_service:/opt/dev/ros2_humble/install/examples_rclcpp_minimal_client:/opt/dev/ros2_humble/install/examples_rclcpp_async_client:/opt/dev/ros2_humble/install/example_interfaces:/opt/dev/ros2_humble/install [... cut ...] ment_package:/opt/dev/ros2_humble/install/ament_mypy:/opt/dev/ros2_humble/install/ament_lint_cmake:/opt/dev/ros2_humble/install/ament_flake8:/opt/dev/ros2_humble/install/ament_copyright:/opt/dev/ros2_humble/install/ament_lint:/opt/dev/ros2_humble/install/ament_index_python:/opt/dev/ros2_humble/install/ament_index_cpp:/opt/dev/ros2_humble/install/ament_cpplint:/opt/dev/ros2_humble/install/ament_cppcheck:/opt/dev/ros2_humble/install/ament_clang_tidy:/opt/dev/ros2_humble/install/ament_clang_format`

minggangw, 2022-10-07:
I ever tried to install the ROS2 Rolling through:
-[Debian packages](https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html)
-[“fat” archive](https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Install-Binary.html)

I haven't built ROS2 from scratch for a long time, but I think the result should be the same with these two above, did you source something like `/opt/ros/rolling/local_setup.zsh`?

MMM-Jasmin, 2022-10-07:
I sourced `local_setup.bash`  or `setup.bash` 

I guess it is not completely the same as when you install it via debian package everything is in /opt/ros/rolling with an include directory following. This include dir does not even exist if you build from source :/
If you build from source there is a subfolder for all ros packages in your install dir.

minggangw, 2022-10-07:
I remember there was once a flag called `--merge-install`, which enables merging all the components together, please check with https://docs.ros.org/en/eloquent/Installation/Linux-Development-Setup.html#build-the-code-in-the-workspace, but not sure whether it's still feasible for Humble. What is certain is that installing from Debian package does merge them together.

MMM-Jasmin, 2022-10-07:
Rebuild ros2 with the `--merge-install` flag and now i also have just the one path in the `$AMENT_PREFIX_PATH` variable.
It should work to install the git version now, but should be mentioned or fixed to work with the not merged version :). 
Nevertheless thanks :)

minggangw, 2022-10-07:
@MMM-Jasmin we will fix this issue soon, thanks!

MMM-Jasmin, 2022-10-07:
just a small typo in 
> (please build wiht flag --merge-install).
but thanks :)

minggangw, 2025-06-30:
Close this issue as no more feedback, please try with latest rclnodejs release to see if it can repro.