RobotWebTools/rclnodejs #888: Error in npm install with Ubuntu 22.04 + Humble

vela-io, 2023-01-05:
**Description**
Hi, everyone. I had this issue when i tried install rclnodejs with npm and source.
 
I have a simple project to library's test and i cant install successfully.  

I had the same result to Ubuntu 20 with foxy.

- Library Version: 0.21.4
- ROS Version: Humble
- Platform / OS: Ubuntu 22.04
- Node version: v18.12.1
- Npm version: 9.2.0

**Steps To Reproduce**
1. source /opt/ros/humble/setup.zsh
2. mkdir rclnodejs-test
3. cd rlcnodejs-test
4. npm init -y
5. npm install rclnodejs

**Expected Behavior**
Install runs without errors

**Actual Behavior**
```
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm ERR! code 1
npm ERR! path /home/rodrigo/rclnodejs-test/node_modules/@rclnodejs/ref-napi
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/rodrigo/.npm/_logs/2023-01-05T22_09_58_874Z-debug-0.log
[error.log](https://github.com/RobotWebTools/rclnodejs/files/10355983/error.log)

```

i've tried update har-validation, uuid and requests but i had same result

samidare210, 2023-01-13:
nodejs version needs to be between  10.23 - 17.x., try to install these version and install the package again

vela-io, 2023-01-13:
Thanks for you return, so the documentation is wrong? 

The prerequisite is written as: "[Nodejs](https://nodejs.org/en/) version between 10.23.1 - 19.x."
And last feature merged include: "Add support for node v18 & v19 https://github.com/RobotWebTools/rclnodejs/pull/884"

Steps was reproduced correctly and lib should install successfully

wayneparrott, 2023-02-06:
@vela-io were you able to make any progress?
I setup a similar env as you described in window wls2. I had an issue with the humble setup.zsh so I reverted to using setup.bash which is my default. With that difference I am able to install rclnodejs with no issues. As a quick test could you try using either node 17 or 19. Note - I recently noticed an issue with node 18.3 but did not replicate the issue with node 18.2 or 18.1. I'll investigate that soon.

kavikode, 2024-03-24:
please try 
npm install rclnodejs@0.22.3

minggangw, 2025-06-30:
Close this issue as no more feedback, please try with latest rclnodejs release to see if it can repro.

closed 2025-06-30 by minggangw