RobotWebTools/rclnodejs #889: README banner github action urls have changed

wayneparrott, 2023-01-10:
The README.md includes the build status for rclnodejs across different ROS2 versions and OS. Those banners are currently broken as outlined here: https://github.com/badges/shields/issues/8671
[image]

minggangw, 2023-01-28:
@wayneparrott thanks for tracking the issue, as I understand, we don't need to take any action, right?

wayneparrott, 2023-01-28:
Yes we have a small change that I plan to submit soon. I'm blocked as I'm not able to test due to #890 Will get this resolved very soon.

minggangw, 2023-08-22:
The actions badges come back now, close this issue. cc @wayneparrott 
[image]

closed 2023-08-22 by minggangw