RobotWebTools/rclnodejs #890: message generation fails on Rolling due to unsupported *.idl files in ros2bag pkg

wayneparrott, 2023-01-27:
The Rolling distro for windows includes the rosbag2_storage_mcap_testdata pkg. In this package there is a the  ComplexMsgDependsOnIdl.msg which includes BasicIdl.idl. The rclnodejs message generator only parses *.msg and *.srv files at this time. Thus the message generator fails with an error. Note that message generation is part of the npm install process for rclnodejs. Thus rclnodejs 0.21.4 can not be used with the Rolling distro on Windows at this time.

wayneparrott, 2023-01-27:
A quick solution is to introduce a mechanism for filtering packages and messages that should not be generated. The `run_test.js` script uses a blocklist.json file to filter test files by OS. I propose we introduce a similar feature for specifying packages and messages to be filtered from the message generation process. Thoughts?

example filter spec:
```
interface MsgGenFilter {
  pkg: Regexp;
  os?: Regexp;
  message?: Regexp;
}
```
Thoughts?

minggangw, 2023-01-28:
Related issue #764

PR #891 (2023-02-24): Custom filter for omitting ROS2 interfaces from the message-generation process
In response to the the blocking issue #890, I've introduced a simple mechanism for filtering ros2 interfaces out of the message generation process. The mechanism reads and external blocklist.json file containing filters that are used by `rosidl_gen/package.js` to determine which ROS2 interface files to excluded during the message-generation process. The user is informed of the ROS interfaces that are omitted with a `console.log` message. 

An ideal solution to #890 is to generate javascript message files from idl fi [... cut]
commit c9b0d096 (2023-02-24): Added RosPackageFilters & blocklist.json (#891)

On windows workflow use node 18.12.0 inplace of 18.13.0. There is a
repeatable issue with node-gyp configuration on node 18.13. It seems to
be related to the node cache. Switching to 18.12 avoids using a cached
version of node 18 and the issue no longer occurs.

Fix #890
commit 06d16465 (2023-04-03): Added RosPackageFilters & blocklist.json (#891)

On windows workflow use node 18.12.0 inplace of 18.13.0. There is a
repeatable issue with node-gyp configuration on node 18.13. It seems to
be related to the node cache. Switching to 18.12 avoids using a cached
version of node 18 and the issue no longer occurs.

Fix #890
closed 2023-02-24 by minggangw