RobotWebTools/rclnodejs #892: Cannot install rclnodejs 0.21.2 and above on Ubuntu 18.04

wujekbrezniew, 2023-02-06:
**Description**

Rclnodejs is not installable on Ubuntu 18.04 with Ros2 Eloquent.

- Library Version: >= 0.21.2
- ROS Version: Eloquent
- Platform / OS: Ubuntu 18.04 Docker container

**Steps To Reproduce**
Do npm i rclnodejs

**Expected Behavior**
C:

**Actual Behavior**

`npm ERR! code 1
npm ERR! path /operator/node_modules/rclnodejs
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! make: Entering directory '/operator/node_modules/rclnodejs/build'
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/addon.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/executor.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/handle_manager.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/rcl_action_bindings.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/rcl_bindings.o
npm ERR! rclnodejs.target.mk:138: recipe for target 'Release/obj.target/rclnodejs/src/rcl_bindings.o' failed
npm ERR! make: Leaving directory '/operator/node_modules/rclnodejs/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@16.13.2 | linux | x64
npm ERR! gyp info find Python using Python version 3.6.9 found at "/usr/bin/python3"
npm ERR! gyp WARN EACCES current user ("auto") does not have permission to access the dev dir "/root/.cache/node-gyp/16.13.2"
npm ERR! gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
npm ERR! (node:58) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
npm ERR! (Use `node --trace-deprecation ...` to show where the warning was created)
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/operator/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/operator/node_modules/rclnodejs/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/operator/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/tmp/.node-gyp/16.13.2/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/tmp/.node-gyp/16.13.2',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/operator/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/tmp/.node-gyp/16.13.2/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/operator/node_modules/rclnodejs',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/rcl_bindings.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE rclnodejs::RclTakeResponse(Nan::NAN_METHOD_ARGS_TYPE)':
npm ERR! ../src/rcl_bindings.cpp:835:3: error: 'rmw_service_info_t' was not declared in this scope
npm ERR!    rmw_service_info_t header;
npm ERR!    ^~~~~~~~~~~~~~~~~~
npm ERR! ../src/rcl_bindings.cpp:835:3: note: suggested alternative: 'rcl_service_init'
npm ERR!    rmw_service_info_t header;
npm ERR!    ^~~~~~~~~~~~~~~~~~
npm ERR!    rcl_service_init
npm ERR! ../src/rcl_bindings.cpp:839:56: error: 'header' was not declared in this scope
npm ERR!    rcl_ret_t ret = rcl_take_response_with_info(client, &header, taken_response);
npm ERR!                                                         ^~~~~~
npm ERR! ../src/rcl_bindings.cpp:839:56: note: suggested alternative: 'h_addr'
npm ERR!    rcl_ret_t ret = rcl_take_response_with_info(client, &header, taken_response);
npm ERR!                                                         ^~~~~~
npm ERR!                                                         h_addr
npm ERR! ../src/rcl_bindings.cpp:839:19: error: 'rcl_take_response_with_info' was not declared in this scope
npm ERR!    rcl_ret_t ret = rcl_take_response_with_info(client, &header, taken_response);
npm ERR!                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! ../src/rcl_bindings.cpp:839:19: note: suggested alternative: 'rcl_take_response'
npm ERR!    rcl_ret_t ret = rcl_take_response_with_info(client, &header, taken_response);
npm ERR!                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR!                    rcl_take_response
npm ERR! At global scope:
npm ERR! cc1plus: warning: unrecognized command line option '-Wno-cast-function-type'
npm ERR! make: *** [Release/obj.target/rclnodejs/src/rcl_bindings.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/operator/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:390:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
npm ERR! gyp ERR! System Linux 5.19.17-051917-generic
npm ERR! gyp ERR! command "/usr/local/lib/nodejs/node-v16.13.2-linux-x64/bin/node" "/operator/node_modules/.bin/node-gyp" "rebuild"
npm ERR! gyp ERR! cwd /operator/node_modules/rclnodejs
npm ERR! gyp ERR! node -v v16.13.2
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2023-02-06T07_30_09_888Z-debug.log
`

wayneparrott, 2023-02-06:
Recent versions of rclnodejs no longer support ROS2 Eloquent. The current level of support is for ROS2 Foxy - the current Rolling version. 

I believe the last version of rclnodejs to support Eloquent is v0.14. But note that the package.json of 0.14 has a maximum node version of 12.X. You can see the Eloquent branch details here https://github.com/RobotWebTools/rclnodejs/blob/eloquent-elusor/package.json

wayneparrott, 2023-02-10:
Interesting - glad you have a working solution. We've focused compatibility testing from Foxy (LTS) forward. We can consider adding a note to the README but I don't see the project investing to extend support and testing back to ROS Eloquent as it is not longer supported by osrf.

minggangw, 2023-06-06:
Close this issue as Eloquent is not supported anymore.

closed 2023-06-06 by minggangw