RobotWebTools/rclnodejs #916: Improving node subscription frequency

gmp-capgemini, 2023-04-20:
This might be more a question than a feature, but I am not getting help anywhere else. I apologize in advance.

I've got a subscription to a camera (`sensor_msgs/msg/Image`) as I am using `rclnodejs` to make a real-time camera display. Even though the topic publishing the camera image is approximately 30 fps, I receive approximately 8 fps. Both publisher and subscriber are on the same host (my laptop), no network involved / no latency involved. It would be nice being able to improve the frequency of received messages for a subscription. I'm just using a basic call to `createSubscription` with an fps_counter (value displayed every second and then resetted):

```
const node_sub = new rclnodejs.Node('subscription_example_node');
  node_sub.createSubscription('sensor_msgs/msg/Image', 'rgb_front/image', (msg) => {
    fps_counter1++;
  });
  node_sub.spin();
```

The link for the [on-line](http://robotwebtools.org/rclnodejs/docs/index.html) documentation is not working. Also executing npm run docs . as suggested by the README of the repo did not work either. I had a view at the implementation of Node.createSubscription and as far as I understood there's an option about QoS, but I could not understand how it works.

Please, lend me a hand with this in case it can be improved.

minggangw, 2023-04-21:
Hi @gmp-capgemini thanks for your feedback, I don't think the bottleneck is the frequency, the current timeout is 10ms and you can try to improve it to see if any change happens:

https://github.com/RobotWebTools/rclnodejs/blob/83958f4a93c2b70bdb51a48a45e5b02cd5821fd5/lib/node.js#L400-L408

The efficiency could be petty low for your case because passing the image data of a frame to the JS side needs to allocate additional memory, which costs lots of time and depends on the image format and size. I would suggest:

1. You can compare with the C++ implementation, e.g., Using [rclcpp](https://github.com/ros2/rclcpp) to implement and see if any improvement there.
2. You should use video streaming instead of `sensor_msgs/msg/Image` if you want the video.

minggangw, 2023-06-06:
@gmp-capgemini  I suggest you can try the latest [0.22.2](https://www.npmjs.com/package/rclnodejs/v/0.22.2?activeTab=versions), which removes the deep copy and may improve the performance.