RobotWebTools/rclnodejs #917: Online version of webdocs is not working

judahtanninen, 2023-04-24:
The url given for the online version of the JSDocs gives a "We can’t connect to the server at robotwebtools.org" error.

Not sure if there is a new location for the docs or not, just figured I should let y'all know.

Url for docs: http://robotwebtools.org/rclnodejs/docs/index.html

wayneparrott, 2023-04-24:
@judahtanninen thx for the reminder for this root issue https://github.com/RobotWebTools/robotwebtools.github.io/issues/82

We will work up an alternative, e.g., github pages or similar solution if the robotwebtools domain is not back online soon.

minggangw, 2023-08-22:
@judahtanninen the doc has been fixed, please try with https://robotwebtools.github.io/rclnodejs/docs/index.html. Close this issue now.

closed 2023-08-22 by minggangw