RobotWebTools/rclnodejs #918: Support Nodejs v20, drop support for v14

wayneparrott, 2023-05-10:
Proposed workflow changes:

1. Nodejs v20 is the current version in development. I propose we modify the workflow to include testing on node v20.

2. Node v14 recently reached end-of-life status. Therefore I propose we drop it from the workflow.

[image]

minggangw, 2023-05-11:
Agree, we can drop node 14 now, thanks!

minggangw, 2023-06-08:
I have updated the documentation for the changes of supported nodejs versions, please check [#17](https://github.com/RobotWebTools/rclnodejs/commit/17a0a7f001bc59c2e85417001f0348490893aa08)

dineshHarikrishnan, 2024-07-29:
> Proposed workflow changes:
>
> 1. Nodejs v20 is the current version in development. I propose we modify the workflow to include testing on node v20.
> 2. Node v14 recently reached end-of-life status. [...]

When can I expect the stable release for the nodeJs version 20?

chfritz, 2024-07-29:
@dineshHarikrishnan we are using node 20 in Transitive (which is installed and running on a lot of robots) and rclnodejs is working fine with it, e.g., in [ros-tool](https://transitiverobotics.com/caps/transitive-robotics/ros-tool/). Is it not working for you?

minggangw, 2025-02-19:
Currently, actions runs on nodejs [`20.X, 22.X, 23.X`](https://github.com/RobotWebTools/rclnodejs/blob/develop/.github/workflows/linux-build-and-test.yml#L33), close this issue.

closed 2025-02-19 by minggangw